steve-community/steve #2: Hubject/OICP integration

goekay, 2015-04-14:


ahzf, 2015-10-13:
Hi, there...

as I'm currently busy with developing a JAVA version of my OICP C# project I would be interested where in your code base I could find s.th. like hooks for:

 - An incoming RFID auth (start) request, in order to dispatch it to Hubject whenever the Id/AuthToken was not found within Steve
- The same for RFID auth stop
- An incoming remote start request
- An incoming remote stop request
- After the charging process is finished: Where to get the trigger and data for sending a charging detail record.

Thanks a lot...
  Achim

goekay, 2015-10-13:
Hello,

-  *An incoming RFID auth (start) request, in order to dispatch it to Hubject whenever the Id/AuthToken was not found within Steve*
[Here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/UserServiceImpl.java#L45) and [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/UserServiceImpl.java#L78) (OCPP version specific)

- *The same for RFID auth stop*
Can you please elaborate what you mean with that? 

- *An incoming remote start/stop request*
Currently, these can only be triggered via the Web page. We do not have an API yet.

-  *After the charging process is finished: Where to get the trigger and data for sending a charging detail record*
The implementation here is once again version specific: [Here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/CentralSystemService12_Service.java#L132) and [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/CentralSystemService15_Service.java#L147). These are messages arriving from the charging station, though, and do not provide a detailed charging record. For this, depending on the wanted data, additional database calls might be necessary to retrieve other related info.

goekay, 2015-10-13:
Can you please inform us about the scope your project? Is it just a translation of the OICP C# project, or a complete OICP integration with Steve? The reason I am asking is... I see you work for GraphDefined and forked Steve. We also started working on the Hubject/OICP integration.

ahzf, 2015-10-13:
Hi... it's not a real project. I just was asked by Hubject to help a little bit, as their documentation is still a bit "sparse", but improving ;) I had a phone call with Dr. Karl-Heinz Krempels a couple of weeks ago and he told me, that it might be a good thing if at least a bit would be ready till this year's eCarTec.

ahzf, 2015-11-27:
Hi again,
at https://github.com/GraphDefined/jOICP you will find a rough Java 1.8+ implementation of OICP v2.0. Perhaps this will help you within your own code base :)

Best from Jena
  Achim

goekay, 2016-06-06:
development/implementation is finished. waiting for approval from hubject to put it online. therefore, we can close the issue.

closed 2016-06-06 by goekay

goRaspy, 2016-06-06:
Great news !
Is the Hubject implementation could works simultaneously with PlugSurfing ?

goekay, 2016-06-06:
no, unfortunately not.

goRaspy, 2016-06-06:
Tks for the precision.

ahzf, 2016-06-06:
@goekay Will it the Hubject connector be fully Open Source? ;)

goekay, 2016-06-06:
depends totally on them. the ball is in their court. the integration principle was the same as with plugsurfing (https://github.com/RWTH-i5-IDSG/steve-plugsurfing). if they give the green light, you will see  another steve repo with a similar approach.

ahzf, 2016-10-06:
Any news on that?

goekay, 2016-10-10:
nope. you should contact and ask hubject, if you want to see any progress.