steve-community/steve #3: WebSocketPingInterval configuration for JSON charge points

goekay, 2015-04-15:
Source: "OCPP implementation guide SOAP - RC1 0.6" and in section "8. Configuration"

Currently ping-pong is done at fixed intervals [1] under the covers without allowing the user to change any configuration [2].

[1] https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/WebSocketConfiguration.java#L37
[2] https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ws/AbstractWebSocketEndpoint.java#L88-L92

lategoodbye, 2019-03-14:
At least it would be nice to make it possible to disable the websocket ping-pong.

goekay, 2019-03-14:
@lategoodbye why do you want to disable it? I am not against the idea, but just want to understand the use case.

lategoodbye, 2019-03-14:
It is common to use OCPP over a mobile connection, which means we should avoid unnecessary traffic at all costs. According to OCPP JSON spec the WebSocket Ping isn't a replacement for the Heartbeat and so considered as not necessary.

Currently i don't plan to use SteVe in such a production environment, but i would be able to test under real condition.

csamsel, 2019-03-15:
The OCPP heartbeat is sent only every couple hours for the stations i know. That time is too long to  prevent a connection being dropped by the CGN-NAT system. And if the connection is dropped it needs to be restablished which is costly in terms of traffic. So preventing the Websocket ping/pong would probably increase the amount of traffic and also might introduce additonal message delay.

goekay, 2019-03-15:
@csamsel heartbeat interval is part of the OCPP "ChangeConfiguration" message which can be set freely by the user. the decision of what is too long or short is the responsibility of the user. by the same token, the spec allows to change or disable the WebSocketPingInterval. if the user wishes to disable it, so be it. but then he/she has to accept possible disadvantageous outcomes like connections drops. it would still be better that we as the provider let the user do whatever he/she wants.

lategoodbye, 2019-03-18:
Hm, according to the spec WebSocketPingInterval only configures the ping on the client side. But i meant the ping on the server side. Sorry for the confusion, should i open a new issue?

goekay, 2019-03-18:
@lategoodbye not necessary. we can track and discuss all websocket ping related issues here. 

ocpp spec does not further specify server-side pings (as is the case with steve). the only sentence i could find is: "... or the server initiates the ping and client responds with Pong."

juherr, 2026-08-25:
Still open, still hard-coded: `PING_INTERVAL = Duration.ofMinutes(15)` — same on `master` and on the `steve-3.14.1` tag.

A use case that didn't exist in the 2019 discussion: **managed reverse proxies**. Putting the OCPP endpoint behind Cloudflare (the same holds for most managed load balancers and ingresses) means idle WebSockets get closed well before 15 minutes, and that timeout is only tunable on Enterprise plans. SteVe's own keep-alive never gets a chance to fire.

The workaround is to drop the OCPP `HeartbeatInterval` below the proxy timeout. It works, but that setting is **global to the installation** — every station pays for it, including the metered mobile ones @lategoodbye wanted to spare, to work around a constraint that applies to one network path.

One configurable value (default 15m, `0` to disable, as @goekay suggested above) covers both sides. Happy to open a PR if the shape is agreed. `IDLE_TIMEOUT` sits right next to it in the same class and is just as fixed — worth deciding whether it belongs in the same change, given #1191.

---
*Drafted by Claude under @juherr's supervision.*

closed 2026-09-07 by goekay