steve-community/steve #7: feature request: sortable user and transaction lists

csamsel, 2015-08-14:
beeing able to sort transactions, e.g., by start time, would be nice.

csamsel, 2015-10-09:
closed by b7335101c92de0eb2eec91c17328f19a74473dac

goekay, 2015-10-09:
let's leave this issue open, since filtering does not substitute for sorting. and some users might still want that feature. following the principle "filter in backend, sort in frontend", we might implement it using http://tablesorter.com/docs/

jiuchongxiao, 2015-11-05:
the commond that "java -jar target/steve-*.jar" didn't work
how should i do to test it?

goekay, 2015-11-05:
did you actually type "java -jar target/steve-*.jar"? 

* is just placeholder for the version number. please replace it with something like 2.0.3

jiuchongxiao, 2015-11-05:
yes,id did actually type it with my version. but it didn't work

jiuchongxiao, 2015-11-05:
[INFO] Replacing /Users/feng/Desktop/ocppDemon/steve/target/steve-2.0.2.jar with /Users/feng/Desktop/ocppDemon/steve/target/steve-2.0.2-shaded.jar
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 11.702 s
[INFO] Finished at: 2015-11-05T19:39:42+08:00
[INFO] Final Memory: 46M/235M
[INFO] ------------------------------------------------------------------------
feng:steve feng$ java -jar target/steve-2.0.2.jar 

it just like above,it didn't work

goRaspy, 2015-11-05:
For information, I just tried with sucess to deploy the new version (2.0.3). 
Be sure that you correctly installed java 8.

Regards

goekay, 2015-11-05:
can you please elaborate what didn't work?

wollodev, 2015-11-05:
Please use stackoverflow or private messages for this kind of conversation. That has nothing to do with this issue. Thanks.

jiuchongxiao, 2015-11-06:
thank you all very much! it work now .

commit 9b21911e (2016-07-21): make data tables sortable by columns (Closes #7)
closed 2016-07-21 by goekay