steve-community/steve #12: Custom Namespace ("ns1") for NKT-Devices // Benutzerdefinierter Namensraum ("ns1") für NKT-Ladesäulen

Nos-, 2015-12-16:
Beim Vergleich der Tests zwischen

1. SteVe <--> Ladesäulensimulator
2. SteVe <--> realer NKT-Ladesäule

ist im Wiresharkmitschnitt aufgefallen, das die NKT-Säule im Gegensatz zur Simulation u.a. den Namespace "*ns1*" (im Head sowie auch im Body) ihres SOAP-Authentifizierungspaketes verwendet.

Im Branch custom-ns-prefix wurden zwar bereits einige dieser Unterschiede angeglichen, allerdings wird der *ns1*-Namespace bisher nur im Header aber noch nicht im Body berücksichtigt.*

Für die weitere Arbeit ist es erforderlich, das dies weiterverfolgt und der spezielle Namensraum überall verwendet wird.

goekay, 2016-01-04:
First of all, let's keep the communication here in English, so that others can also benefit. Secondly, the problem is not due to a *custom namespace*, because OCPP has clearly defined namespaces that must be used in messages. And our SOAP framework, CXF, always sets the namespace. Always. What you are talking about is the usage of prefixes in the message body. 

In particular, it's a problem because your NKT charging station expects a custom *prefix* to be used. It seems, that the developers hardcoded the prefix in the software, which makes the software broken as it is totally unimportant what the prefix is, as long as the namespaces are set. Please contact the manufacturer to report this issue.

That being said, the workaround is trivial and analogous to the previous work in 5ab05f5b6910cd74f37f4fca173a1fc246e3c211, only this time the namespaces contain "Cs" instead of "Cp".

closed 2016-01-04 by goekay

Nos-, 2016-01-08:
Many Thanks!

It seems that I do not See the wood for the trees.

commit d33fd4ad (2016-01-04): Closes #12