steve-community/steve #14: steve-2.0.5 : feature to bulk insert of ocpp tags and charge points

goRaspy, 2016-03-02:
Hello,

Could you detail how works this feature ?

Regards

goekay, 2016-03-02:
for example, if you are on the page to add a charge point, above the "Add Charge Point" portion of the page there is a bar named "Add Charge Point List". this bar is clickable. if you click on it, the form to input ChargeBox IDs will present itself.  now, in this form we need a delimiter to separate charge points. and this is the line break. therefore, you should enter a ChargeBox ID per line.

since it's a bulk insert feature, you cannot specify other fields as in the single insert form. but still if you wish, you can enter additional details in the details/update pages for each charge point afterwards.

does this answer your question?

goRaspy, 2016-03-02:
Many thanks it is very clear.

I just tried with success to add more than 100 OCPP tags.

closed 2016-03-02 by goRaspy