steve-community/steve #22: Stove value update at each meter values

goRaspy, 2016-04-19:
For active transaction, in transaction list. 
It should be great to update the "stop value field" by the latest meter value. 

So we will have a view of delivered power during the transaction.

goekay, 2016-04-19:
to display the meter values sent for a transaction is a pending TODO for me actually.

but to realize it the way you described (by setting the stop value field) would be confusing, because

1) from a semantic standpoint "stop value" should be reserved for final value as the name suggests, and not for in-between values, and
2) our internal logic depends on stop timestamp and/or stop value being set, when trying to figure out whether a transaction is finished. having a value set might break some logic. we do not want that.

we are already writing these values into database ("connector_meter_value" table), so the information is never lost. these are just not presented to the user, yet.

instead, we can make the transaction ids on the transaction page clickable. upon click we read these entries from the table for a transaction and you are directed to another view which lists all the meter value entries. this way, we would be able to list the in-between meter values not only for active transactions, but also for finished transactions, which should be interesting for some users.

goRaspy, 2016-04-19:
Hi, I think your idea is very good. A possibility to view old meters values should very interesting. In a future we may also have a power graph/curb for each transaction.

commit 1322c23c (2016-04-27): Closes #22

add "transaction details" page which lists intermediate meter values
closed 2016-04-27 by goekay