steve-community/steve #24: jooq.steve.db.routines does not exist

Gingonic, 2016-06-22:
Hi there,

I cannot build the latest master version of Steve.
I'm trying to build it with the maven:onbuild docker image on docker-machine with a windows client.

I get: 

```
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.3:compile (default-compile) on project steve: Compilation failure
[ERROR] /usr/src/app/src/main/java/de/rwth/idsg/steve/repository/impl/GenericRepositoryImpl.java:[7,30] package jooq.steve.db.routines does not exist
```

Any thoughts?

goekay, 2016-06-23:
did not test it with docker/windows, but we can compile it.

my guess is your database user does not have the required privileges for jooq to read the meta information about tables and routines from database. please make sure that you applied the first step in the configuration: https://github.com/RWTH-i5-IDSG/steve#configuration-and-installation

Gingonic, 2016-06-23:
I did. It seems to be a missing package and don't think the database itself is the error, those are pretty clear. Then again I'm not a java guy.

```
package jooq.steve.db.routines does not exist
```

goekay, 2016-06-23:
this package will be generated by jooq during build. so, first of all, it's not an external package/dependency. this package is needed for a stored procedure that we use (get_stats). can you make sure that this procedure is created?

additionally, can you post your maven build log here?

csamsel, 2016-06-24:
The error is covered in the FAQ:

https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-can-i-backup--transfer-the-steve-database

goekay, 2016-07-18:
since the issuer did not respond for a while, i presume his problem is resolved, and close this issue.

closed 2016-07-18 by goekay

alestmej, 2018-02-22:
I had the same problem and here is the solution:

1. run:  _mvn package_  - it generates some files into directory: target\generated-sources
2. in your IDE mark the directory _target\generated-sources_ as "source directory"

now you can build&run from your IDE (main class to execute is _de.rwth.idsg.steve.Application_).

goekay, 2018-02-22:
thanks for the update. it is probably helpful for people having trouble when running steve fom IDE, but @Gingonic was trying to build the app within docker.