steve-community/steve #25: Store and lock unknow user from past start transaction

goRaspy, 2016-07-14:
When a charging station is offline, it is better for user experience to accept any occp tag. But when the Steve connection is back, the charging station send to the back-office the list of "offline" transactions.

So when a transaction was started by a not registered ocpp tag, the charging station send recurly this transaction which is not accepted by Steve. In this case we need a manual creation of the tag or a clear cache of the charging station.
It should be great if Steve automatically add the tag in database and just after the lock it.

goekay, 2016-07-14:
why does the charging station accept unregistered occp tags for a transaction start when it's offline? as fas as i know, the charging station keeps a white list of used ocpp tags and will use these when it's offline. so, the charging station should not just allow unknown tags for charging.

goRaspy, 2016-07-14:
In public area, due to range anxiety of EV driver. I think that the white list is not enough for offline charging station. By definition, public charging station are here for "outside home" drivers. So the user do not care about network problem, are he is not "known" by the charge point white list (and really have to charge).

In my point of view, with the actual state of emobility market it is more fair to let everybody charge in case in technical problem than close (partially) the charging station.

csamsel, 2016-07-15:
Hard to believe that any operator would configure a charging station in a way that you can simply use a GSM/UMTS jammer to charge for free.

goRaspy, 2016-07-15:
I did not though at this possibilies. It is interesting ;)

By the way, I am sure that many changing station manufacturer deliver stations setuped like that. With roaming networks, the white list system is not enought.  

By the way, the case do not happen every day.

goekay, 2016-07-15:
if you are using an ocpp 1.5 charging station you can basically send every ocpp tag known to steve to charging station (by using Send Local List operation). this will update the white list of the charging station. doesn't this solve your problem?

goekay, 2016-07-15:
if the operator really wants to provide charging to every one for free, why use ocpp at all? make the charging station stand-alone and simple like a public power outlet (without any intelligence) so every one can use it.

goekay, 2016-07-18:
in ocpp, there are many concepts based on id tags: authorization, transaction/reservation identification, local authorization list in charging station (white list), syncing the auth lists etc. so the protocol itself considers important to keep track of _who does what_.

therefore, steve cannot accept any, unknown, unregistered id tags. i will not implement this feature request.

closed 2016-07-18 by goekay

goRaspy, 2016-07-22:
Hi 

Sorry gor the answer delay,

Many thanks for your precision. I understand your point of view and I did not think about GSM jammer possibilities. Complete network leak are not happenning often, so it is not a real problem. I will manage white lists, but with an external data source like plugsurfing. In case of network leak the charging station will be unavailable. Maybe in a future (not a priority) it should be interesting to send the information of unavailability to plugsurfing if the charging station lost connection.

goekay, 2016-07-22:
> Maybe in a future (not a priority) it should be interesting to send the information of unavailability to plugsurfing if the charging station lost connection.

this is relatively easy to realize. we can do that.

goekay, 2016-07-22:
>> Maybe in a future (not a priority) it should be interesting to send the information of unavailability to plugsurfing if the charging station lost connection.

> this is relatively easy to realize. we can do that.

my last commit in steve-plugsurfing repository implements this feature.

goRaspy, 2016-07-22:
I'll try it asap ! many thanks.

goRaspy, 2016-07-26:
Hi, I made somes tests with the new status update in plugsurfing.  The link works fine and very fast.  But unfortunately,  this update cause more problems than solve. 

When an "old" JSON session is close by timeout.  Steve send an "inactive"  status to ps,  and do not take in consideration if a more recent session is active.  So for each reboot,  the old JSON session  is close after a while and Ps received an inactive status.  No matter if the station is well connected.  
In this case steve should check if another session is active before update the status. 

There is the similar  problem for mail alert,  but it is it not major problem because it doesn't change  anything for the final user. 

Regards 

There is the same problem for th e

goekay, 2016-07-26:
thanks for the report, both issues should be resolved now with the latest commits

goRaspy, 2016-07-26:
Awesome ! Seems works perfectly. Will give you a report after longer testing.
(I will also open an issue in steve-ps for a small problem)