steve-community/steve #26: [feature] desactivate automatic redirection from root of serveur

goRaspy, 2016-08-17:
I would like try to run several instance of the Steve running on the same server (different JAR and database).
In my vision, each JAR will be configured with a different context path with the objective to have different endpoint on the same IP/PORT.

For the moment, steve automatically redirect connection from server root to => serveur:port/context/manager. I think it should be a problem to run several instance of Steve simultaneously. I also suppose that should be easy to remove this automatic redirection. 

I there a way to simply desactivate this redirection ?

goekay, 2016-08-17:
the problem is not the redirection. even if we deactivate redirection, you still cannot achieve what you want. you can start multiple instances with the same ip, but not with the same port. 

steve comes with an embedded web application server (jetty). when you start steve, we read parameters from file (ip/port) and first start jetty with these parameters. and if the port that you try to use is used by another application (read: another steve instance), jetty will fail to start, ergo, steve will fail to start.

if you want, you can leave ip and context path the same, and change the ip for each steve instance, and do it like that.

the redirection itself i find useful because you do not have to type the whole path to go to the main page of steve.

historical info:
at the beginning, steve was just a web application and did not contain an embedded server. the idea was to start a separate web application server (tomcat) and deploy the web app to the server (read: copy the war file to the correct tomcat directory). with such a setup, you were able to start one server, and run multiple steve instances with different context paths within the server, since server is decoupled from applications. but this introduced some complexity in the configuration (you had to configure 1) tomcat and 2) steve in different places), and i was receiving questions how to set everything up. the users were not familiar with tomcat and did not know how to do it properly. for this reason, i decided to embed the server into steve, do the basic configuration in the code myself, and have all the possible parameters in one place (read: main.properties) to make it easier for the users. so, i do not want to go back to how it was at the beginning.

closed 2016-08-17 by goekay

goRaspy, 2016-08-17:
Hi, tanks for the explanation. I will make some new tries. Thanks for your
suggestion.

You tell "if you want, you can leave ip and context path the same, and
change the ip for each steve instance, and do it like that."
But I think you would tell "change the *PORT* for each steve instance".

With an external config file it should be very easy to manage.

Concerning historical. I confirm you, my firsts test with Steve was not
easy due to tomcat configuration ^^

Many thanks

goekay, 2016-08-17:
yes. that's right. i meant **port** and not the **ip**.