steve-community/steve #27: [question] External config file

goRaspy, 2016-08-17:
During compilation of Steve we have a message "The Java system property '{}' can be set to point to an external properties file". Is that would tell that Steve could run with external main.properties file ?

It should be great to be able to change parameters without recompiling the file.

goekay, 2016-08-17:
the actual log message is: 

`Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one`

yes it is there for external properties file. "main.properties" ist the name of the setting (java system property). you can set it to any value without compiling. i think you are mixing the _name_ of the setting with the _value_ of the setting.

for how to use it please see the steve-2.1.0 release details: [https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-2.1.0](https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-2.1.0)

closed 2016-08-17 by goekay

goRaspy, 2016-08-17:
Many thanks it is perfect. Sorry for my bad English :(

I will check if it could works with a relative link for the path of the external_file.properties.