steve-community/steve #29: OCPP 1.6 compatibility

aziz-khalapoum, 2016-12-07:
Hello,
I all-ready tried Steve, and it is pretty awesome. I have a question concerning OCPP 1.6 compatibility.

OCPP 1.6 is very near from OCPP 1.5-J. New features concern mainly smart-charging. As specified in protocol definition, only "core" features are mandatory and we allready have them in Steve (1.5).

So, I would like to know if it is possible to update Steve and let it accept OCCP 1.6 charging stations and consider them like a 1.5(-J) charging station.

I am pretty sure there is something to do in /src/main/java/de/rwth/idsg/steve/ocpp/OcppVersion.java but my tests did not work.

Regards

goekay, 2016-12-09:
thank you for your interest. even though there might be few functional differences between OCPP 1.5 and OCPP 1.6, there will be much more going on in the concrete implementation. it's not even close to changing a few lines here and there (like in OcppVersion). unfortunately, there will be a lot of "code/boilerplate ceremony" and lots of little things to be considered. potential db schema changes, web ui changes to trigger operations in version 1.6 stations (this will be huge), etc.

because of this and since we do not have a pressing demand for the new version, for the time being i will postpone the new version implementation. having said that, you are welcome (or anyone from the community) to undertake this task and we can gladly help/assist you.

goRaspy, 2017-03-21:
Hi, I am also interested to find a solution to let the "handshake" between existing Steve and 1.6 charge-points works. 
I clearly understand that will not allow "new" features "works" (like smart-charging) but for the moment most of charge point manufacturers just implemented "mandatory" features witch finally is pretty the same thing of 1.5-J.

MathijsStertefeld, 2017-05-17:
Hey goekay,

I am also interested in implementening SteVe for our system. However, we are still deciding whether to host SteVe ourselves, or use an active provider. Support for 1.6 is a big issue in this case, since we want to be future proof and have support for it. Has any progress been made towards implementing support for it yet, and if not, is there any interest for it? Otherwise, we might consider the cost of implementing it ourselves, or consider an alternative instead.

goekay, 2017-05-18:
hello, there has not been any work done regarding the 1.6 implementation. there are multiple reasons for this. between other activities and projects that we are busy with, it's hard to find and dedicate time for it. at the end, steve started as a side project and is not at the core of my/our day-to-day operations. moreover, outside this github issue, we did not receive any formal requests from industrial partners. i do not want to say that we will _never_ do it, it's just hard to say _when_.

on the other hand, you can be assured that any possible issue/problem/bug report w.r.t the current implementation will be resolved asap, because it's more important to have no issues with the current implementation than to extend it (with new features i.e. OCPP 1.6 impl).

ktr87, 2018-02-20:
Hello Goekay,

I have gone through the previous discussion regarding the compatibility of steve with OCPP 1.6 version.
I have checked the message formats of 1.5j and 1.6, There are slight changes in 1.6 compared to 1.5j
 
I am attaching 1.5j and 1.6 comparison file. Please check it. I want to make Steve work with 1.6. 
Let me know what the class files required to change in order steve to work with 1.6 basic functionality like, bootNotification, statusNotification, start/stopTransaction, meterValues, heatBeat, etc. 
I would appreciate your support on this.

[comparision_1.6_1.5j.txt](https://github.com/RWTH-i5-IDSG/steve/files/1739002/comparision_1.6_1.5j.txt)

goekay, 2018-03-12:
so... we are preparing the codebase for the ocpp 1.6 implementation. for the last few days, we started to unify classes and logic (where generalisation is possible) in order to prevent logic being distributed to many places. our aim is to have as few as possible version-specific classes, such that the changes needed for ocpp 1.6 are bearable (and the number probable mistakes is reduced).

our internal tests show that everything is working, but still... i kindly ask you to test (in a non-production environment!) that these changes did not introduce some bugs/problems and report to me if that is the case. if there are no problems, then we can start implementing ocpp 1.6.

JavaIsJavaScript, 2018-03-13:
Hey goekay,

I've been working on implementing OCPP 1.6 in SteVe aswell using the old Class Structure.
[https://github.com/JavaIsJavaScript/steve-1.6](url)

I have successfully implemented all features of 1.5 for 1.6 as I believe.

I'm kind of stuck on the Set Charging Profile feature that is new in OCPP 1.6 which gives me a "TypeConstraintViolation" even when i'm using the example values that are given in the OCPP 1.6 Specification on page 28.

and as @ktr87 mentioned there are slight changes between 1.5 and 1.6. Basically the only difference that really occures to SteVe as i found is when a Charge Point calls MeterValues.req

---------

And for bugs that I found in the current version of SteVe, I submitted a Pull Request for UpdateFirmware since you were able to call it with the absence of the "retrieveDate" parameter and this parameter is mandatory as described in the OCPP specifiaction. (apparently a check failed).

Aswell as a "java.lang.NullPointerException" when performing a "SendLocalList.req".

goekay, 2018-03-13:
@JavaIsJavaScript can you make a pull request for NullPointerException please?

JavaIsJavaScript, 2018-03-13:
Working on it.

goekay, 2018-03-14:
after the commit 85f2000f7b05557a4bd1b5faf79f13dfac60fb16 the ocpp 1.6 service (charging station -> backend) implementation is complete both for websocket/json and soap. you can test it. the ocpp 1.6 client (backend -> charging station) needs some more work (mainly web ui changes) to trigger operations.

on the other hand, we will at first only implement the reservation feature profile (as defined in the ocpp 1.6 spec). so smart charging and remote trigger will be left out.

aziz-khalapoum, 2018-03-14:
Hello,

Great for wanting to upgrade Steve. I wanted to test it but I have a compilation error : 

/steve/src/main/java/de/rwth/idsg/steve/web/dto/ocpp/UpdateFirmwareParams.java:[31,6] cannot find symbo

 Any help will be welcome.

JavaIsJavaScript, 2018-03-14:
@aziz-khalapoum the class is missing this import "import javax.validation.constraints.NotNull;"

goekay, 2018-03-14:
@aziz-khalapoum you are using the code in [master](https://github.com/RWTH-i5-IDSG/steve/tree/master) branch, whereas the ocpp 1.6 implementation happens in [ocpp-1.6-impl](https://github.com/RWTH-i5-IDSG/steve/tree/ocpp-1.6-impl) branch. ocpp-1.6-impl does not contain the PR #37 which was broken due to missing import (fixed with PR #41). but they will be after after finishing the ocpp 1.6 implementation and merging with master.

aziz-khalapoum, 2018-03-14:
Thank you. I am doing the tests and I will send you the feedback.

Many thanks

aziz-khalapoum, 2018-03-14:
I just performed the tests with a physical charging station.
The following commands work very well:
BootNotification
Authorize
StartTransaction
StopTransaction

I am impatient to test the other commands when the operation interface is implemented (Ocpp 1.6)

Thanks

goekay, 2018-03-17:
the client implementation of ocpp 1.6 is done. you can test it.

ktr87, 2018-03-18:
Hello Goekay,
Good work and thanks for the update. I haven't taken the updated code yet. But,  with little modifications to your previous update for 1.5, i am able to test the following commands with real ocpp1.6 version charge point.
BootNotification
Authorize
Status Notification
StartTransaction
StopTransaction
MeterValues

Anyway, i will take updates and test it again.

aziz-khalapoum, 2018-03-19:
Hello @goekay ,

Great, I just tested the new update. The commands I had to test are:
Start Transaction;
Stop Transaction;
Remote Start;
Remote Stop;
Change Availability;
reset;
Get Configuration.

All these commands are working good except the reservation. 
Once the reservation is made, all the other badges are blocked but after passing the badge used to make the reservation, it generates an error on the logs. 
Otherwise, from the graphical interface, everything works perfectly.

You can find attached the logs in the following mail. 

Thanks

goekay, 2018-03-19:
@aziz-khalapoum can you please test it after my last commits?

aziz-khalapoum, 2018-03-19:
Hello @goekay ,
Perfectly,  it works very well. I am really happy to have solved this problem.

Many Thanks

goekay, 2018-03-20:
thanks to @JavaIsJavaScript TriggerMessage operation is also implemented. if you cannot find more issues with the implementation, i will merge the dev branch with master and will make a release (finally, after a long time).

goekay, 2018-03-20:
Closed by d008b89100e4419cc6670c1fba38d0c0fd995028

closed 2018-03-20 by goekay