steve-community/steve #30: OCPP 1.5 Reservation

aziz-khalapoum, 2017-03-27:
Hello,

I just tested the reservation This works well for the first occurrence of the reserved badge (OCPP tag). Once the reserved badge has passed, the point charge remains blocked and no other badge passes or even this booked. The reservation is on the dashboard of the administration but I can not cancel it. So I had to shut down the application and delete the reservation at the database. And once restarted. To conclude from what I understood is that once the badge reserved is accepted, steve fails to close (canceled) the reservation.

If you need the logs, let me know.

Any help will be welcome.
Thank you

goekay, 2017-03-28:
hello, can you please send me the logs?

aziz-khalapoum, 2017-03-28:
Hello, 
I just sent you the logs to your email.
Thanks

goekay, 2017-03-28:
thanks for the logs. after looking at them, it seems to me that you have network errors and a big problem related to database connectivity. all the exceptions hint at that and not at the transaction/reservation/whatever logic that is implemented in steve. examples:

- HikariPool-1 - Connection com.mysql.jdbc.JDBC4Connection marked as broken because of SQLSTATE(08S01), ErrorCode(0)
- Communications link failure - The last packet successfully received from the server was 335,225 milliseconds ago.  The last packet sent successfully to the server was 335,224 milliseconds ago.
- java.net.ConnectException: Connection refused (Connection refused)
- Error getting connection from data source HikariDataSource (HikariPool-1) - Connection is not available, request timed out after 30000ms.

for instance, charging station websocket connection attempts fail with "org.springframework.web.socket.server.HandshakeFailureException" because in every attempt we try to look up the chargeBoxId in the database to verify that the charging station is registered. and that fails, because the connection pool implementation (HikariPool) cannot get a database connection from the pool. 

so, i think, the issue is not about the reservation. i cannot rule out the possibility that there might be something wrong with our implementation, but in order to identify and diagnose that problem first the other components should work properly. please take a look at your database and network configuration.

closed 2017-03-28 by goekay

sahar1987, 2019-05-14:
Hi everyone,
Has the reservation worked well ?I mean is there any proof of concept for reservation using SteVe? or Is it still under development?
your reply is  appreciated

goRaspy, 2019-05-14:
Works fine for me ( ocpp 1.5 )