steve-community/steve #31: HTTPS implement

aziz-khalapoum, 2017-10-04:
Hello,

I would like to know what are the steps and the configuration to make to set up the https. I saw that the configuration file foresees it : 
# Jetty HTTPS configuration
#
https.enabled = false
https.port = 8443
keystore.path =
keystore.password =

thank you

goekay, 2017-10-05:
hello, well first you have to enable https and change the port (if you wish). the main part, though, is configuring the container for certificates for java applications with the java keytool. there are many tutorials that you can google and find, but here is one to get you started: https://www.digitalocean.com/community/tutorials/java-keytool-essentials-working-with-java-keystores

aziz-khalapoum, 2017-10-05:
hello, 
I was able to make it work.Thank you for your help.

closed 2017-10-05 by aziz-khalapoum

josecharry, 2020-09-01:
Thanks for your answers. I implemented https creating a repository of keys with the following cmd: _keytool -genkey -alias **test** -keyalg RSA -keystore **test.jks** -keysize 2048_ , then you have to put the path and the pass in main.properties.

rdc-Green, 2021-02-03:
How do you find the path @josecharry to enter into main.properties after entering the cmd?

rdc-Green, 2021-02-03:
I found it in /home/user/test.jks