steve-community/steve #34: maximumPoolSize in Hikari

goRaspy, 2017-11-27:
In some case, when Steve is stopped (or crashed), when you restart it and if there if more than 10 charging station reconnecting simultaneously, database access can crash (again) with HikariPoolError.

It seems that can be cause by the  maximumPoolSize of Hikari parameters which is by default at 10 and can be enlarge with the spring.datasource.hikari.maximum-pool-size parameters ( exemple https://github.com/jhipster/generator-jhipster/issues/3150).

goekay, 2017-11-27:
it can indeed be a connection pool size problem, but when we are talking about 10 database access attempts and not hundreds or some bigger number like that, it would surprise me that the database cannot handle it with default connection pool size, assuming your server resources are not low and hardware is somewhat up-to-date. 

can you send me 1) the logs please so that i can have a deeper look into the issue and 2) your hardware configuration?

goRaspy, 2017-11-27:
Ok I will try to reproduce the problem to have logs. To be honest it happened with more than 10 connected stations ( approximately 25) and  another case with a PlugSurfing's connection leak ( we entered into a loop and Steve created many "new transactions" with the IDTag until the HiraliPoolError).

Concerning hardware, we use dedicated Debian server with free resources (CPU and RAM). So I do not thing this is the problem. But it may be checked next time I'll get the problem. 

It it snot easy for me to reproduce the problem without generate issues on working charging station. I'll keep you in touch.

Thanks for you fast answer.

goekay, 2017-12-03:
since i didn't hear from you for a week, i am closing this issue. i think, the issue you mention is not related to steve or connection pool size, anyway. a deeper inspection should reveal that the source of the problem lies somewhere else. but, nevertheless, if you want to increase the  pool size, you can use the method setMaximumPoolSize(...) of HikariConfig to some desired value after initializing it in https://github.com/RWTH-i5-IDSG/steve/blob/d95ea0a2b57fa1ce2106a6ece1e4b28f3b33c2f3/src/main/java/de/rwth/idsg/steve/config/BeanConfiguration.java#L77

closed 2017-12-03 by goekay