steve-community/steve #35: Showing Errors in all the java files

ktr87, 2018-02-19:
Hi,
I just imported 'steve' into my eclipse workspace and changed properties file params.
I am able to BUILD the project successfully and application is working fine. 

But in eclipse, almost every java file showing errors. 
for example: in main class Apllication.java i am getting the following errors
-- "log" cannot be resolved 
-- The method "getProfile()" is undefined for the type SteveConfiguration
How to resolve these errors?
I appreciate your support.

goekay, 2018-02-19:
hello,

we are using [lombok](https://projectlombok.org/) to auto-generate some boilerplate code like getters, setters, etc. by using annotations. it is a compiler hook that replaces the annotations with respective code. that's why you can build the project, and everything is working fine.

however, for IDEs (be it intellij idea, eclipse or something else) you need a plugin since they are working in a different environment. for eclipse: https://projectlombok.org/setup/eclipse

ktr87, 2018-02-19:
Hello Goekay,

Thank you very much for the quick response.
Now, all errors are gone.

I forgot to mention one more error in pom.xml
I am getting the following error twice in the <executions> blocks migrate and generate.
  
--Plugin execution not covered by lifecycle configuration: org.jooq:jooq-codegen-maven:3.8.3:generate (execution: default, phase: generate-sources)

But i am ignoring this error in eclipse build. does it have any effect on the application?

goekay, 2018-02-19:
it seems like an eclipse problem: https://stackoverflow.com/questions/6352208/how-to-solve-plugin-execution-not-covered-by-lifecycle-configuration-for-sprin?rq=1

bad maven support of eclipse (buggy m2e plugin) is the reason i switched from eclipse to intellij idea.

closed 2018-02-20 by goekay