steve-community/steve #38: NullPointerException for Send Local List

JavaIsJavaScript, 2018-03-13:
Haven't found a fix yet with the new class structure @goekay hence this issue.

NullPointerException when calling SendLocalList.

`[ERROR] 2018-03-13 14:51:31,862 de.rwth.idsg.steve.web.GlobalControllerAdvice - Request: http://127.0.0.1:8080/steve/manager/operations/v1.5/SendLocalList raised following exception.

java.lang.NullPointerException: null
	at de.rwth.idsg.steve.utils.StringUtils.getOperationName(StringUtils.java:21) ~[steve-2.1.0.jar:?]
	at de.rwth.idsg.steve.ocpp.CommunicationTask.<init>(CommunicationTask.java:78) ~[steve-2.1.0.jar:?]
	at de.rwth.idsg.steve.ocpp.CommunicationTask.<init>(CommunicationTask.java:55) ~[steve-2.1.0.jar:?]
	at de.rwth.idsg.steve.ocpp.task.SendLocalListTask.<init>(SendLocalListTask.java:28) ~[steve-2.1.0.jar:?]
	at de.rwth.idsg.steve.service.ChargePointService15_Client.sendLocalList(ChargePointService15_Client.java:95) ~[steve-2.1.0.jar:?]
	at de.rwth.idsg.steve.web.controller.Ocpp15Controller.postSendList(Ocpp15Controller.java:368) ~[steve-2.1.0.jar:?]`

goekay, 2018-03-13:
do you see other exceptions in the logs?

goekay, 2018-03-13:
the only reason why it can happen is, that the object SendLocalListRequest is not created and therefore is null. it can be related to the database calls that are necessary for creating this object.

closed 2018-03-13 by goekay