steve-community/steve #44: OCPP 1.6 "Smart Charging" Profile

goekay, 2018-03-20:
due to the extra overhead, this profile is left out for release [3.0.0](https://github.com/RWTH-i5-IDSG/steve/releases/tag/steve-3.0.0). this issue keeps track of the implementation status and any related discussion.

JavaIsJavaScript, 2018-03-20:
In the OCPP specification at SetChargingProfile.req it says: If connectorId = 0,
the message contains an overall limit for the Charge Point.

and at GetCompositeSchedule.req it says: When connectorId=0, the Charge Point will calculate
the expected consumption for the grid connection

even though the connectorId is required at both are these also usable on multiple stations at once?

goekay, 2018-03-20:
i think so. i did not look at the spec in detail. but... if the user wants to set the same charging profile for, lets say, connectorId=3 of multiple stations, so be it. if he means the whole station for all the stations, then you can set connectorId=0. i think this case is similar to ChangeAvailability operation. you can look into its implementation.

however, i have to say one thing: i think these smart charging operations are meant for some tight integration with other operations and automated action/reaction. therefore, a straightforward implementation having a web page with a html form where you can set some values and send a message to charging station(s) will not be that much valuable. since i am not in the business of selling/using/maintaining charging stations i do not have the necessary use cases to implement them correctly. that's why i left them out in the first place.

i think smart charging profile is little bit tricky to correctly implement.

JavaIsJavaScript, 2018-03-20:
The only tricky one is SetChargingProfile as i noticed because it can be related to transactions. but you can set it outside of a transaction as well which would work with multiple i guess.

and GetCompositeSchedule would just return Rejected for some Charge Points if for example the connectorId is unknown.

But anyway thanks, I'll look into it.

goekay, 2018-03-20:
yeah, but the data model Charging Profile is quite complicated and the user cannot (should not) define profiles on-the-fly, send and forget them. 

the most useful implementation would be to have a "charging profile management" view, where you can define your profiles 1a) for all stations 1b) specific for a station 2a) for all users 2b) specific for a user and _store_ these in the database. since the data model is complicated, for better UX we should have a GUI like [wlan on/off timeplan for routers](https://welzmueller.de/wp-content/uploads/2015/08/WLAN-Zeitschaltung_nach-zeitplan-abschalten.jpg), that allows easy modification with mouse.

for example, ClearChargingProfile has an optional "stackLevel" field, which is set in SetChargingProfile. so we have to keep track of charging profiles (an indication for storing profiles in database).

another example is the response to GetCompositeSchedule. the ChargingSchedule objects are so complex, that you cannot read a status value in print in a table anymore (as it mostly was). this is an indication that StringOcppCallbacks will not work.

all i am saying is, the _smart charging profile is little bit tricky to implement._

chuck-h, 2018-03-25:
Definitely interested in seeing this (OCPP 1.6 local smart charging) for a real-world installation we are planning at my cohousing group. Goal is load management (to avoid exceeding electrical service capacity). Charge points may be EV-Box business line, but decision is not yet made. Starting with 4 stations, final build-out approx 25 stations.

V2G-UK, 2018-04-16:
We also have an interest in getting the OCPP 1.6 "smart charging" feature working. However SteVe worked "out of the box" for us:

[http://www.V2G-EVSE.com/2018/04/16/ev-charging-station-administration-using-ocpp-1-6-with-steve/
](http://www.V2G-EVSE.com/2018/04/16/ev-charging-station-administration-using-ocpp-1-6-with-steve/
)
That means we haven't even needed to glance at the source code yet. Where do you recommend that we start?

JavaIsJavaScript, 2018-04-16:
@V2G-UK ,

I have been working on the "Smart Charging" profile of OCPP 1.6 aswell and have implemented all of the functions in my fork. But it is far from perfect, since for example the Charging Profiles that can be set at Charge Points don't get saved in the Database and thus you won't be able to retrieve them and see which profiles were set on specific Charge Points.

The functions SetChargingProfile, ClearChargingProfile and GetCompositeSchedule do work, atleast in a testing environment.

V2G-UK, 2018-04-16:
Thanks very much @JavaIsJavaScript - I'll take a good look when I have a spare moment, which I suspect won't be for a few days.

chuck-h, 2018-05-13:
I'm just starting to exercise OCPP1.6J smart charging with a Delta AC Mini (firmware v04.04.30.03). I find that I can set a TxDefault profile, but cannot read back with GetCompositeSchedule as I expected. Can anyone tell me whether my expectation is wrong, or maybe the charge station is misbehaving?

_SetChargingProfile_ exchange:  
[2,"7d67f2c9-61c6-4be7-8677-ef805c6f9854","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":100,"stackLevel":0,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Recurring","recurrencyKind":"Daily","validFrom":"2018-05-12T00:00:00.000Z","validTo":"2018-05-19T00:00:00.000Z","chargingSchedule":{"duration":86400,"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":25.5,"numberPhases":1},{"startPeriod":28800,"limit":15.5,"numberPhases":1},{"startPeriod":72000,"limit":25,"numberPhases":1}]}}}]

[3,"7d67f2c9-61c6-4be7-8677-ef805c6f9854",{"status":"Accepted"}]

_GetCompositeSchedule_ exchange:  
[2,"1a457d15-729c-454a-a276-61eb5f2b2dbd","GetCompositeSchedule",{"connectorId":1,"duration":86400,"chargingRateUnit":"A"}]

[3,"1a457d15-729c-454a-a276-61eb5f2b2dbd",{"status":"Rejected"}]

_P.S._: I'm using the fork from @JavaIsJavaScript, hope it's okay to post on this issue here.

goekay, 2018-05-14:
it seems like the station is misbehaving.

goekay, 2018-11-20:
i merged the branch [ocpp-1.6-smart-charging](https://github.com/RWTH-i5-IDSG/steve/tree/ocpp-1.6-smart-charging) into master and therefore consider this issue resolved.

closed 2018-11-20 by goekay