steve-community/steve #45: Error Message does not match the OCPP 1.6 specification.

nuno407, 2018-03-21:
Hi,
When the server send a CALLERROR the message does not match the specification.

Received: [4,"UID2077486715","InternalError","Internal services failed while processing of the payload","SQL [insert into `stevedb`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timesta..."]

the entry "SQL [insert into `stevedb`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timesta..."" should be a object and is a string.
JSON Schecma of CALLERROR:

    {
    "$schema": "http://json-schema.org/draft-04/schema#",
    "title": "CallError",
    "type": "object",
    "properties": {
        "errorCode": {
            "type": "string",
            "enum": [
                "NotImplemented",
                "NotSupported",
                "InternalError",
                "ProtocolError",
                "SecurityError",
                "FormationViolation",
                "PropertyConstraintViolation",
                "OccurenceConstraintViolation",
                "TypeConstraintViolation",
                "GenericError"
            ]
        },
        "errorDescription": {
            "type": "string",
            "maxLength": 20
        },
        "errorDetails": {
            "type": "object"
        }
    },
    "additionalProperties": false,
    "required": [
        "messageTypeId",
        "idTag",
        "errorCode",
        "ErrorDescription",
        "errorDetails"
    ]
    }

Thank you, 
nuno

goekay, 2018-03-21:
working on it. can you tell me what caused the exception? maybe there is an issue with handling of meter values?

JavaIsJavaScript, 2018-03-21:
I stumbled across this whilst testing MeterValues:

 [4,"jKg4ShEDt4PxSv3J4291tioh2ES2K3dV","InternalError","Internal services failed while processing of the payload","SQL [insert into
 `stevedb`.`connector_meter_value` (`connector_pk`, `transaction_pk`, `value_timesta..."]

But this was because i was using a transactionId that was not related to the Charge Point, after using a correct transactionId related to the Charge Point it just worked perfectly.

goekay, 2018-03-21:
alright.

nuno407, 2018-03-21:
Thank you @JavaIsJavaScript.
That's correct!
Issue should be closed for now.

goekay, 2018-03-21:
> This "InternalError" happens aswell if the transactionId sent by the Charge Point in the MeterValues.req does not exist

@JavaIsJavaScript are you sure about this? During my tests MeterValues.req without a transaction id do not cause an internal error.

JavaIsJavaScript, 2018-03-21:
Yeah, without transactionId works, but if the transactionId in the request doesn't exist in the database i ment

goekay, 2018-03-21:
ah okay, then it's not our problem :)

closed 2018-03-21 by goekay

nuno407, 2018-03-21:
But @goekay,
The error response still does not match the specification. In this case a "PropertyConstraintViolation" as defined in OCPP-J v1.6 should be throw right?
Thank you for the help,
nuno

goekay, 2018-03-21:
technically speaking, you are right. but the fact, whether a transaction id exists or not, is checked within the database (foreign key constraint). if it does not exist, we get in the application a chain of exceptions: SQLIntegrityConstraintViolationException in BatchUpdateException in DataAccessException. at this point, it's really hard to find out what went why wrong, other than looking at the error string returned by the database.

this is one of the reasons, why we cannot use some of the finer grained error codes.

goekay, 2018-03-21:
> EDIT: As well as, the entry "errorDetails" should be an object and not a string as the server returns.

@nuno407 i know. my commit fixed this and only this, as i explained in my previous comment

nuno407, 2018-03-21:
Thank you @goekay !