steve-community/steve #48: OCPP IDTags expiry date does not match the configuration in the Steve portal

nuno407, 2018-03-26:
I have the following OCPP ID Tags:
[image]
And the  response is the following:

    Received: [2,"1f762142-f659-4a2a-a278-5148f00921b3","SendLocalList",{"listVersion":9,"localAuthorizationList":[{"idTag":"user","idTagInfo":{"expiryDate":"2018-03-26T16:44:14.689Z","status":"Accepted"}},{"idTag":"user1","idTagInfo":{"expiryDate":"2018-03-26T16:44:14.689Z","status":"Accepted"}},{"idTag":"user2","idTagInfo":{"expiryDate":"2018-03-26T16:44:14.689Z","status":"Accepted"}}],"updateType":"Full"}]

What could be the problem?
Thank you,
Nuno

goekay, 2018-03-26:
are you living in a country with utc+1 timezone?

nuno407, 2018-03-26:
Yes. But the expiry date year does not match anyway

goekay, 2018-03-26:
the thing is, steve, the java application, is set to use UTC as time zone. if your database's timezone varies from that, during the reading/writing such results might occur.

goekay, 2018-03-26:
on the other hand, the LocalAuthorizationList message is not wrong. "16:44:14.689Z" denotes the time in UTC (because of "Z" as suffix), which translates to 17:43 (after rounding) in your time zone. so.. i don't get what the issue is here?

nuno407, 2018-03-26:
Ok. Then the solution is to use UTC timezone in the database also?
Thank you for the help.

goekay, 2018-03-26:
as a best practice: steve time zone == mysql time zone

but i do not see any problem with the LocalAuthorizationList mesasage.

goekay, 2018-03-26:
ah, i know what you mean, the date sent in LocalAuthorizationList does not match the actual expiry date from database. this was to prevent a charging station to read from local/cached white list all the time, if for example the expiry is 5 years from now, and trigger it to ask Steve for the actual, up-to-date authorization status. in the settings tab, you can see an Expiration setting. it controls the time window for caching local id tags.

goekay, 2018-03-26:
maybe it's time to rethink whether this limiting/strict behaviour is necessary...

closed 2018-03-26 by goekay

nuno407, 2018-03-27:
Hi @goekay,
Sorry for the not so fast response. But Yeap I think you fixed the issue. 
Thank you! :smiley:

commit 8ad33a98 (2018-03-26): loosen id tag expiration constraints (Closes #48)

reason: until now, we always used current time + some hours (set in UI)
for id tag expiration, regardless of an expiry date the database had
for this id tag. the idea behind was to better control two possible
situations:
- no expiry date: set an explicit one to prevent the station caching it,
such that after some time, it _has_ to ask Steve.
- expiry date too far in the future: since business rules, contracts,
conditions and therefore expiration of id tags might change, we wanted
to prevent the stations from storing id tags for too long in the [... cut]