steve-community/steve #53: Chargepoint 1.6 simulator

victormunoz, 2018-04-05:
I would like to test the new OCPP 1.6 features with Steve with some chargepoint simulator.
What simulator do you recommend?

I have tried with the ChargeTimeEU chargepoint (https://github.com/ChargeTimeEU/Java-OCA-OCPP), that implements version 1.6, but haven't succeeded...

JavaIsJavaScript, 2018-04-05:
@victormunoz  i got a hacky simulator that works on 1.6 on my github, its not perfect but it works for testing, atleast for me

goekay, 2018-04-05:
@victormunoz why didn't ChargeTimeEU work? As far as i know, they only provide the skeletons for json/soap communication and the user/developer has to provide/implement content for the message fields.

goekay, 2018-04-05:
some time ago we experimented with https://github.com/NewMotion/ocpp-charger and https://github.com/aymen-mouelhi/ocpp-js. they were okay.

JavaIsJavaScript, 2018-04-05:
I was using this one http://www.gir.fr/ocppjs/ for 1.5 testing and just changed and added some things so it works for 1.6

TVolden, 2018-04-05:
That's correct @goekay. I limited the lib to the specifications from OCA.

victormunoz, 2018-04-05:
@JavaIsJavaScript I have tried your project. Looks good!
Although the "npm install" seemed to fail (many errors in the console), the program apparently runs fine.
It outputs the message "Connected to Central System", and the steve log shows "New connection established".
Should it make something else? I expected some BootNotification, heartbeat, meter values...
Anyway, good work! Thank you!

JavaIsJavaScript, 2018-04-05:
The errors are known, type bootnotification in the console and press enter, you can do this with basically al functions from cp to cs.

goekay, 2018-04-05:
steve actually has the building blocks to simulate chargepoints in test directory. please look at these exemplary implementations, which you can modify as you wish:
- [for soap](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/test/java/de/rwth/idsg/steve/ApplicationTest.java)
- [for json](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/test/java/de/rwth/idsg/steve/ApplicationJsonTest.java)

you create your request objects, set their fields, send them to steve, and do whatever you like with the response (e.g. verify the fields in response).

nenecmrf, 2018-04-12:
Started working on a really simple CP Simulator, simply using html, js, websockets
It already can do send requests to the server and handle server requests.
Would be happy if you try to implement something to it.
Link https://github.com/nenecmrf/OCPP-J-CP-Simulator

mctsnt, 2020-11-16:
I have tried testing SteVe with the NewMotion docile charge point simulator(https://github.com/NewMotion/ocpp) using OCPP 1.6J.

I have managed to get successful heartbeat and authorization 
[image]

 but SteVe cannot detect the version of OCPP I am using so I am unable to do any other operation such as get diagnostics:
[image]

When I then  try to start the operation via the charge point simulator I get an error message saying the expected message was not received.
[image]

Is this the expected sequence of events, should I expect to see OCPP version in the picture before being able to proceed?

 

Is there another client (charging point) simulator that is more up-to-date or compatible with SteVe that you can recommend?

goekay, 2025-09-16:
some other discoveries:

- https://github.com/SAP/e-mobility-charging-stations-simulator
- https://github.com/matth-x/MicroOcppSimulator