steve-community/steve #54: Database link error

BobOntario, 2018-04-10:
I followed the readme to install the packege. When I ran "mvn package", got the below messages:

[INFO] Scanning for projects...
[INFO] 
[INFO] -------------------------< de.rwth.idsg:steve >-------------------------
[INFO] Building steve 3.0.1
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- properties-maven-plugin:1.0.0:read-project-properties (default) @ steve ---
[INFO] 
[INFO] --- cxf-codegen-plugin:3.2.4:wsdl2java (generate-sources) @ steve ---
[INFO] 
[INFO] --- flyway-maven-plugin:4.2.0:migrate (default) @ steve ---
Loading class `com.mysql.jdbc.Driver'. This is deprecated. The new driver class is `com.mysql.cj.jdbc.Driver'. The driver is automatically registered via the SPI and manual loading of the driver class is generally unnecessary.
[INFO] Flyway 4.2.0 by Boxfuse
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 2.264 s
[INFO] Finished at: 2018-04-10T09:54:40+08:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:4.2.0:migrate (default) on project steve: org.flywaydb.core.internal.dbsupport.FlywaySqlException: 
[ERROR] Unable to obtain Jdbc connection from DataSource (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] -----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 08S01
[ERROR] Error Code : 0
[ERROR] Message    : Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] : Path does not chain with any of the trust anchors
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException

Can someone give some help what's wrong with this?

Thanks

JavaIsJavaScript, 2018-04-10:
did you do the database preperation? and you need to change the credentials in the [main.properties](src/main/resources/config/prod/main.properties)

goekay, 2018-04-10:
as @JavaIsJavaScript said, please make sure that the database is configured and the credentials in main.properties are set correctly. in addition to that (if it is really a bug) can you please specify your operating system (distro, version), java version any mysql/mariadb version?

goekay, 2018-04-11:
so what was the problem? how did you solve it?

BobOntario, 2018-04-11:
Sorry, the issue still there. The above is software version I used. Any information is highly appreciated.

csamsel, 2018-04-11:
You might have to explicitly activate that mysql binds to the port by commenting skip-networking like so:

```
# skip-networking
bind-address                            = 127.0.0.1
```

BobOntario, 2018-04-11:
is there any requirement on JDBC driver?

BobOntario, 2018-04-12:
Finally I can make "mvn package" successful.    Thanks.
But I got an error when I run "Java -jar target/steve-3.0.1.jar"

root@ecs-3929-2:/steve-master# java -jar target/steve-3.0.1.jar, please see below. Please help!!!  If need log file, I can upload it.

Log file: /root/logs/steve.log
Starting........... FAILED!
Please refer to the log file for details
Exception in thread "main" java.net.BindException: Cannot assign requested address
	at sun.nio.ch.Net.bind0(Native Method)
	at sun.nio.ch.Net.bind(Net.java:433)
	at sun.nio.ch.Net.bind(Net.java:425)
	at sun.nio.ch.ServerSocketChannelImpl.bind(ServerSocketChannelImpl.java:223)
	at sun.nio.ch.ServerSocketAdaptor.bind(ServerSocketAdaptor.java:74)
	at org.eclipse.jetty.server.ServerConnector.openAcceptChannel(ServerConnector.java:340)
	at org.eclipse.jetty.server.ServerConnector.open(ServerConnector.java:308)
	at org.eclipse.jetty.server.AbstractNetworkConnector.doStart(AbstractNetworkConnector.java:80)
	at org.eclipse.jetty.server.ServerConnector.doStart(ServerConnector.java:244)
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68)
	at org.eclipse.jetty.server.Server.doStart(Server.java:398)
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68)
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:135)
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:37)
	at de.rwth.idsg.steve.Application.start(Application.java:44)
	at de.rwth.idsg.steve.Application.main(Application.java:38)

[image]

BobOntario, 2018-04-12:
[INFO ] 2018-04-12 09:52:13,740 de.rwth.idsg.steve.utils.PropertiesFileLoader - Hint: The Java system property 'main.properties' can be set to point to an external properties file, which will be prioritized over the bundled one
[INFO ] 2018-04-12 09:52:13,748 de.rwth.idsg.steve.Application - Loaded the properties. Starting with the 'PROD' profile
[INFO ] 2018-04-12 09:52:13,769 de.rwth.idsg.steve.Application - Date/time zone of the application is set to UTC. Current date/time: 2018-04-12T01:52:13.755Z
[INFO ] 2018-04-12 09:52:13,792 org.eclipse.jetty.util.log - Logging initialized @600ms to org.eclipse.jetty.util.log.Slf4jLog
[INFO ] 2018-04-12 09:52:14,297 org.eclipse.jetty.server.Server - jetty-9.4.z-SNAPSHOT; built: 2018-03-20T12:21:10Z; git: 1f8159b1e4a42d3f79997021ea1609f2fbac6de5; jvm 1.8.0_161-b12
[INFO ] 2018-04-12 09:52:14,399 org.eclipse.jetty.server.session - DefaultSessionIdManager workerName=node0
[INFO ] 2018-04-12 09:52:14,399 org.eclipse.jetty.server.session - No SessionScavenger set, using defaults
[INFO ] 2018-04-12 09:52:14,400 org.eclipse.jetty.server.session - Scavenging every 600000ms
[INFO ] 2018-04-12 09:52:14,406 org.eclipse.jetty.server.handler.ContextHandler.steve - Initializing Spring root WebApplicationContext
[INFO ] 2018-04-12 09:52:14,406 org.springframework.web.context.ContextLoader - Root WebApplicationContext: initialization started
[INFO ] 2018-04-12 09:52:14,426 org.springframework.web.context.support.AnnotationConfigWebApplicationContext - Refreshing Root WebApplicationContext: startup date [Thu Apr 12 01:52:14 UTC 2018]; root of context hierarchy
[INFO ] 2018-04-12 09:52:14,469 org.springframework.web.context.support.AnnotationConfigWebApplicationContext - Scanning base packages: [de.rwth.idsg.steve.config]
[INFO ] 2018-04-12 09:52:15,178 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Starting...
[INFO ] 2018-04-12 09:52:15,508 com.zaxxer.hikari.HikariDataSource - HikariPool-1 - Start completed.
[INFO ] 2018-04-12 09:52:16,285 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2018-04-12 09:52:16,856 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP12
[INFO ] 2018-04-12 09:52:16,872 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2012/06/}CentralSystemService from class ocpp.cs._2012._06.CentralSystemService
[INFO ] 2018-04-12 09:52:16,963 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP15
[INFO ] 2018-04-12 09:52:16,964 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2015/10/}CentralSystemService from class ocpp.cs._2015._10.CentralSystemService
[INFO ] 2018-04-12 09:52:17,040 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemServiceOCPP16
[INFO ] 2018-04-12 09:52:17,054 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cs/2010/08/}CentralSystemService from class ocpp.cs._2010._08.CentralSystemService
[INFO ] 2018-04-12 09:52:17,061 org.apache.cxf.endpoint.ServerImpl - Setting the server's publish address to be /CentralSystemService
[INFO ] 2018-04-12 09:52:19,341 org.springframework.web.socket.server.support.WebSocketHandlerMapping - Mapped URL path [/websocket/CentralSystemService/*] onto handler of type [class org.springframework.web.socket.server.support.WebSocketHttpRequestHandler]
[INFO ] 2018-04-12 09:52:19,492 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern='/static/**'], []
[INFO ] 2018-04-12 09:52:19,492 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: Ant [pattern='/views/**'], []
[INFO ] 2018-04-12 09:52:19,554 org.springframework.security.web.DefaultSecurityFilterChain - Creating filter chain: org.springframework.security.web.util.matcher.AnyRequestMatcher@1, [org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter@51b87df7, org.springframework.security.web.context.SecurityContextPersistenceFilter@1fe8f5e8, org.springframework.security.web.header.HeaderWriterFilter@7b297740, org.springframework.security.web.csrf.CsrfFilter@7dbae40, org.springframework.security.web.authentication.logout.LogoutFilter@621f23ac, org.springframework.security.web.authentication.UsernamePasswordAuthenticationFilter@77c66a4f, org.springframework.security.web.authentication.www.BasicAuthenticationFilter@615439f7, org.springframework.security.web.savedrequest.RequestCacheAwareFilter@6e3dd5ce, org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter@ef86c72, org.springframework.security.web.authentication.AnonymousAuthenticationFilter@2315052d, org.springframework.security.web.session.SessionManagementFilter@57ce2898, org.springframework.security.web.access.ExceptionTranslationFilter@1c7f9861, org.springframework.security.web.access.intercept.FilterSecurityInterceptor@45bbc52f]
[INFO ] 2018-04-12 09:52:19,645 org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerMapping - Mapped "{[/manager/home/ocppJsonStatus],methods=[GET]}" onto public java.lang.String de.rwth.idsg.steve.web.controller.HomeController.getOcppJsonStatus(org.springframework.ui.Model)
[... cut ...]
	at org.eclipse.jetty.server.Server.doStart(Server.java:398) ~[steve-3.0.1.jar:?]
	at org.eclipse.jetty.util.component.AbstractLifeCycle.start(AbstractLifeCycle.java:68) ~[steve-3.0.1.jar:?]
	at de.rwth.idsg.steve.JettyServer.start(JettyServer.java:135) ~[steve-3.0.1.jar:?]
	at de.rwth.idsg.steve.SteveProdStarter.start(SteveProdStarter.java:37) ~[steve-3.0.1.jar:?]
	at de.rwth.idsg.steve.Application.start(Application.java:44) ~[steve-3.0.1.jar:?]
	at de.rwth.idsg.steve.Application.main(Application.java:38) ~[steve-3.0.1.jar:?]

csamsel, 2018-04-12:
The error message is clear, the set ip:port combination in main.properties is wrong. Either the IP does not belong to the machine or the port is not usable because of permissions or because it's already in use.

goekay, 2018-04-12:
actually, if the address is already in use, then the exception message states this with "Address already in use". so you should look for the source of the problem somewhere else.

BobOntario, 2018-04-12:
thanks everyone. it's works!!!!

closed 2018-04-12 by BobOntario

sukhjitsehra, 2019-01-17:
I am getting the following error while running mvn package: 

[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:5.1.4:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: 
[ERROR] Unable to obtain connection from database (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 08S01
[ERROR] Error Code : 0
[ERROR] Message    : Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.

*** Can you suggest the issue ***

goekay, 2019-01-17:
@sukhjitsehra we are tracking this issue in #104. i am posting this just for completeness and future visitors of this issue.