steve-community/steve #56: API features (OCPI ?)

goRaspy, 2018-04-11:
I do love steve-plugsurfing for "OIOI hacks" possibilities. It is very precious to be able to play remotely with charging stations by the plugsurfing's API.
Unfortunately, I do not think it is possible to do a "simple" merge of steve 1.6 on steve-plugsurfing (1.5).
In the same time OIOI evolves to v4. So steve endpoint should also need to be upgraded.

In my opinion, the best "API" to integrate should be OCPI which seems to become the new industry standard. However this option involves probably a "structural" evolution of steve to integrate emi3 denominations for EVSE.

goekay, 2018-04-11:
neither merging steve 1.6 with steve-plugsurfing nor upgrading to OIOI v4 is on the horizon.

i agree, that OCPI started to become the new industry standard, but cannot say if or when i will implement it. maybe in the really distant future....

rdc-Green, 2022-10-28:
@goRaspy  did you manage to implement OCPI with SteVe?

juherr, 2022-10-30:
@rdc-Green managing OCPI and managing OCPP are 2 different jobs. 
With @goekay adding an API (see #910) and some more notifications (like proposed in #847), it will be easy to start a 3rd party OCPI gateway using Steve as an OCCP server.

pir8radio, 2023-07-21:
i came looking for ocpi abilities as well.  main reason being able to tie my stations to public map services like plugshare.com so users can see if chargers are available and remotely monitor their charging status if they leave their car at my charger.    Would be nice to see this, would love to tie it to plugshare.com