steve-community/steve #57: Connection is not available, request timed out after 30002ms

aziz-khalapoum, 2018-04-18:
Hello, 
I notice these exceptions for last few days.

```
Caused by: java.sql.SQLTransientConnectionException: HikariPool-1 -
Connection is not available, request timed out after 30000ms.
    at com.zaxxer.hikari.pool.HikariPool.createTimeoutException(HikariPool.java:555)
    at com.zaxxer.hikari.pool.HikariPool.getConnection(HikariPool.java:188)
    at com.zaxxer.hikari.pool.HikariPool.getConnection(HikariPool.java:147)
    at com.zaxxer.hikari.HikariDataSource.getConnection(HikariDataSource.java:83)
    at org.jooq.impl.DataSourceConnectionProvider.acquire(DataSourceConnectionProvider.java:86)
    ... 36 more
```

I saw an open issue on HikariCP ((https://github.com/brettwooldridge/HikariCP/issues/1007))since we use the same library.

any help is welcomed

goekay, 2018-04-18:
how recent is the exception? after my commit a8bb23f2c9fb0ef062457f6926882b4c0d0e807f, which updates hikari version?

aziz-khalapoum, 2018-04-18:
I used the old version. I will update and see if the problem will be solved.
I will keep you informed quickly.

Many Thanks

aziz-khalapoum, 2018-04-19:
I’m still using the 2.1.0 version (since I need plugsurfing). As the code has been refactored, the merge isn’t that obvious. So here are the changes I made on the concerned file (BeanConfiguration and pom.xml). the compilation went well as well as the execution. If my changes are good, could you give me your feedback. I will be happy if there is a commit for the release 2.1.0 : 

```
private void initDataSource() {

      SteveConfiguration.DB dbConfig = CONFIG.getDb();

      HikariConfig hc = new HikariConfig();

      hc.setJdbcUrl("jdbc:mysql://" + dbConfig.getIp() + ":" + dbConfig.getPort() + "/" + dbConfig.getSchema());
      hc.setUsername(dbConfig.getUserName());
      hc.setPassword(dbConfig.getPassword());

      hc.addDataSourceProperty("cachePrepStmts", "true");
      hc.addDataSourceProperty("prepStmtCacheSize", "250");
      hc.addDataSourceProperty("prepStmtCacheSqlLimit", "2048");

          // we must explicitly set the encoding, otherwise data is not handled properly!
      hc.addDataSourceProperty("encoding", "utf8");
      hc.addDataSourceProperty("useUnicode", "true");

      dataSource = new HikariDataSource(hc);
    }
```

Many thanks

goekay, 2018-04-19:
to be frank, i do not think that this issue is because of hikaricp or its version. when you opened the issue, for a moment i thought that it might be related to it since we recently updated it.

you only posted one part of the exception stack trace, so i don't know what actually caused it, but i think this was a problem that existed in steve's code for a long time. only recently we started to test steve under heavy load: first with https://github.com/chargegrid/abusive-charge-point and since yesterday [with our own impl](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/test/java/de/rwth/idsg/steve/StressTestSoapOCPP16.java). after testing with them we discovered some bottle necks in the code (almost all related to database operations), which we _hopefully_ fixed. i assume your version suffers from the same problems and bottlenecks.

therefore you should look at the recent commits starting with 69eb2fc00264bb454611b4806ae57cc71cce4b4f for a solution.

goekay, 2018-04-19:
> I will be happy if there is a commit for the release 2.1.0

we do not plan to continue supporting plugsurfing version anymore.

closed 2018-05-07 by goekay