steve-community/steve #58: Question: SOAP OCPP endpoint

lnjuanj, 2018-04-27:
Hi there,
I have been implementing a 1.5 SOAP client (no WS), and all communications initiated by the client to steve went perfect!
However, now I wanted to implement the server-initiated ones. I went into the Operations section but my client was not there. I then looked into the steve code and I think I found those operations are only available for websockets. Is this correct? If that were not the case, how could I introduce the client endpoint? (in the client details, in steve dashboard, the endpoint appears empty).
Thanks in advance!

csamsel, 2018-04-27:
No, all 1.5 operations are implemented for both SOAP and JSON over Websockets. You'll the to register your client as chargepoint and also send heartbeats as a real charging station would do.

goekay, 2018-04-27:
@lnjuanj what operations are you talking about? as you can see [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/ocpp/ChargePointService15_InvokerImpl.java) all operations are implemented for both transports. can you illustrate your problem in more detail?

goekay, 2018-04-27:
> how could I introduce the client endpoint? 

ocpp soap spec uses [ws-adressing](https://en.wikipedia.org/wiki/WS-Addressing) and the ws-adressing header "from" should be the url/endpoint of the station. you should set it. every incoming soap message is intercepted by steve and if the header "from" is set, the endpoint adresss in database is updated. 

so... in summary: look into ocpp spec, your soap client must implement ws-adressing.

lnjuanj, 2018-04-27:
Hi! I didn't know I had to introduce the "From" header in the ws-addressing. In fact, I think I had read some info about ws in the code and had thought it mean websockets... 😢 
Now, my problem is introducing the "From" header, since the "To" is introduced easily through the WSDL, but I need to figure out this other tag.
Thanks everyone!

goekay, 2018-04-27:
@lnjuanj i think we solved the mystery here, and therefore i am closing this issue. i hope it's okay for you.

closed 2018-04-27 by goekay