steve-community/steve #63: How to keep System running after close Terminal?

BobOntario, 2018-05-11:
Hello, team

Sorry to bother you. We are using Steve system, which is running in cloud. But we found that we have to restart it every time after we close terminal.  Wondering if we can make it as a service...even we close the terminal, the system still running...

Thanks

JavaIsJavaScript, 2018-05-11:
Maybe by using the "screen" command? https://www.digitalocean.com/community/tutorials/how-to-install-and-use-screen-on-an-ubuntu-cloud-server

BobOntario, 2018-05-11:
Thanks, but not sure how "screen" can support my expectation...

csamsel, 2018-05-11:
screen would work.

you can also do:
`java -jar target/steve.jar &`

goekay, 2018-05-11:
in addition to what's being written, if you want to start steve every time after startup automatically (cloud providers sometimes restart instances due to maintanece) you could prepare a shell script that inclues `java -jar target/steve.jar` and put it either in `/etc/init.d` or use `crontab` as explained [here](https://stackoverflow.com/questions/12973777/how-to-run-a-shell-script-at-startup) or [here](http://www.jcgonzalez.com/linux-java-service-wrapper-example). this way steve will always run in background and survive vm restarts.

BobOntario, 2018-05-11:
Thanks

Sent from my Bell Samsung device over Canada's largest network.

closed 2018-05-13 by goekay