steve-community/steve #68: Accessing steve web interface from local network

ghamkhari, 2018-05-24:
Currently,  steve web interface is only accessible from the local host:

`http://<your-server-ip>:<port>/steve/manager`

I need to access the web server from another computer in the same local network (LAN). What configurations should I change?

ghamkhari, 2018-05-24:
`server.host` in this [this](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L22) file should be set to `0:0:0:0`

goekay, 2018-05-24:
this is kind of explained [here](https://github.com/RWTH-i5-IDSG/steve/wiki/Configuration#multiple-interfaces). using `0.0.0.0` implies steve is reachable from _all_ network interfaces, which might not be desirable. as an alternative, you can find out the ip adress that the server gets in the local network and use it in `server.host`. another option is [runing steve behind a reverse proxy](https://github.com/RWTH-i5-IDSG/steve/wiki/FAQ#how-can-i-run-steve-behind-a-reverse-proxy) (for example, nginx).