steve-community/steve #70: Authorization and Authentication for Rest API in Steve

ktr87, 2018-05-29:
I am tying to test the steve with Restful API to get the list of charge points and transactions. For this, I have implemented Rest API in steve. But i am not able to authenticate and authorize the API's.

Created the following classes for authentication and authorization:
1. StevInitializer extends AbstractAnnotationConfigDispatcherServletInitializer
2. CustomBasicsAuthenticationEntryPoint extends BasicAuthenticationEntryPoint
3. StevFilter implement Filter

The container is not loading the StevInitializer class We tried to make changes in StevAppContext.java but no luck.

Please need your help on this.

goekay, 2018-05-29:
without looking at what you have done, it's really hard to guess what the problem is. where is your project/implementation?

ktr87, 2018-05-29:
Hi @goekay ,

I am attaching my class files. Please have a look.

I have a written a  rest controller class and the following method. 
class level : @RequestMapping(value = "/Services")

@RequestMapping(value = "/getAllChargePoints", method = RequestMethod.GET,produces = MediaType.APPLICATION_JSON_VALUE)
    public ResponseEntity<?> getAllCS() {
    	List<ChargePointsResponse> cbs = chargePointRepository.getAllChargePoints();
        return new ResponseEntity<List<ChargePointsResponse>>(cbs,HttpStatus.OK );
    }

I have created my own "Users" table and added username and password in the table.

Currently, I am able to get all the list of charge points without any security. But, i want to add Authorization and Authentication to the rest call. How can i implement this in Steve?

I appreciate your support on this. 

Thanks.

[myclasses.zip](https://github.com/RWTH-i5-IDSG/steve/files/2048895/myclasses.zip)

goekay, 2018-05-29:
well, the classes you provided do not do anything useful, actually. 

> Currently, I am able to get all the list of charge points without any security.

this is because your api mapping `"/Services" ` does not use the `"/manager"` prefix (which is ok as far as this is the intent and you know why). as you can see [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/config/SecurityConfiguration.java#L53-L56), only paths starting with this prefix are protected.

> But, i want to add Authorization and Authentication to the rest call. How can i implement this in Steve?

this is possible. since steve uses the spring framework, there are sub projects of spring that deal with rest and security with oauth and you can integrate them into steve. there are numerous online tutorials and guides, but as a starting point you can look at these: [1](https://spring.io/guides/tutorials/bookmarks/#_building_a_rest_service), [2](https://spring.io/guides/tutorials/bookmarks/#_securing_a_rest_service), [3](http://projects.spring.io/spring-security-oauth/docs/oauth2.html).

however, this issue has very little to do with steve and a lot to do with spring and how it works. i cannot help you further.

ktr87, 2018-05-29:
Okay. Thanks. I will work on this.

closed 2018-05-30 by goekay

TechOverflow, 2018-08-09:
Has anyone got this working and maybe have tips on doing so?