steve-community/steve #72: Deadlock found when trying to get lock; try restarting transaction

JakobRb, 2018-06-27:
Hello,
I've found in some circumstances, a chargebox is sending duplicate StopTransaction message i a short period of time (milliseconds). This situation ends with a MySQL error "Deadlock found when trying to get lock; try restarting transaction"
Example from today:
```
[INFO ] 2018-06-27 04:12:53,713 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: a6be55ac-152b-40e2-91a2-97d7892f7840
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">XXXXXXXXXXXXXXXX</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:c1a2a3df-061a-4fb67-a479-b8aee6234cd1</wsa5:MessageID><wsa5:From><wsa5:Address>http://XXX.XXX.XXX.XXX:XXXXX</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://XXX.com:8080/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StopTransaction</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:stopTransactionRequest><cs:transactionId>9999</cs:transactionId><cs:idTag>XXXXXXXX</cs:idTag><cs:timestamp>2018-06-27T04:10:10Z</cs:timestamp><cs:meterStop>99999</cs:meterStop></cs:stopTransactionRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[INFO ] 2018-06-27 04:12:53,724 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: a6be55ac-152b-40e2-91a2-97d7892f7840
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">XXXXXXXXXXXXXXXX</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:c1a2a3df-061a-4fb67-a479-b8aee6234cd1</wsa5:MessageID><wsa5:From><wsa5:Address>http://XXX.XXX.XXX.XXX:XXXXX</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://XXX.com:8080/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StopTransaction</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:stopTransactionRequest><cs:transactionId>9999</cs:transactionId><cs:idTag>XXXXXXXX</cs:idTag><cs:timestamp>2018-06-27T04:10:10Z</cs:timestamp><cs:meterStop>99999</cs:meterStop></cs:stopTransactionRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[WARN ] 2018-06-27 04:12:53,724 org.apache.cxf.phase.PhaseInterceptorChain - Application {urn://Ocpp/Cs/2012/06/}CentralSystemService#{urn://Ocpp/Cs/2012/06/}StopTransaction has thrown exception, unwinding now
org.apache.cxf.interceptor.Fault: SQL [update `transaction` set `transaction`.`stop_timestamp` = ?, `transaction`.`stop_value` = ?, `transaction`.`stop_reason` = ? where `transaction`.`transaction_pk` = ?]; Deadlock found when trying to get lock; try restarting transaction
	at org.apache.cxf.service.invoker.AbstractInvoker.createFault(AbstractInvoker.java:162) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.jaxws.AbstractJAXWSMethodInvoker.createFault(AbstractJAXWSMethodInvoker.java:267) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.service.invoker.AbstractInvoker.invoke(AbstractInvoker.java:128) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.jaxws.AbstractJAXWSMethodInvoker.invoke(AbstractJAXWSMethodInvoker.java:232) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.jaxws.JAXWSMethodInvoker.invoke(JAXWSMethodInvoker.java:85) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.service.invoker.AbstractInvoker.invoke(AbstractInvoker.java:74) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor$1.run(ServiceInvokerInterceptor.java:59) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) ~[?:1.8.0_121]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[?:1.8.0_121]
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor$2.run(ServiceInvokerInterceptor.java:126) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.workqueue.SynchronousExecutor.execute(SynchronousExecutor.java:37) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.interceptor.ServiceInvokerInterceptor.handleMessage(ServiceInvokerInterceptor.java:131) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.phase.PhaseInterceptorChain.doIntercept(PhaseInterceptorChain.java:308) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.ChainInitiationObserver.onMessage(ChainInitiationObserver.java:121) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:85) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.MediatorInInterceptor.handleMessage(MediatorInInterceptor.java:36) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.phase.PhaseInterceptorChain.doIntercept(PhaseInterceptorChain.java:308) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.ChainInitiationObserver.onMessage(ChainInitiationObserver.java:121) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.http.AbstractHTTPDestination.invoke(AbstractHTTPDestination.java:267) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invokeDestination(ServletController.java:234) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:208) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.ServletController.invoke(ServletController.java:160) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.CXFNonSpringServlet.invoke(CXFNonSpringServlet.java:216) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.handleRequest(AbstractHTTPServlet.java:301) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.doPost(AbstractHTTPServlet.java:220) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.transport.servlet.AbstractHTTPServlet.service(AbstractHTTPServlet.java:276) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:864) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:535) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:146) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:548) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:257) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1595) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:255) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1253) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:203) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:473) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1564) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:201) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1155) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:144) ~[steve-3.0.2.manager-0-24-0.jar:?]
[... cut ...]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_121]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_121]
	at org.apache.cxf.service.invoker.AbstractInvoker.performInvocation(AbstractInvoker.java:179) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.jaxws.JAXWSMethodInvoker.performInvocation(JAXWSMethodInvoker.java:66) ~[steve-3.0.2.manager-0-24-0.jar:?]
	at org.apache.cxf.service.invoker.AbstractInvoker.invoke(AbstractInvoker.java:96) ~[steve-3.0.2.manager-0-24-0.jar:?]
	... 55 more
```

goekay, 2018-06-27:
which steve version are you using? more specifically, what is the git commit id of your installation?

JakobRb, 2018-06-27:
Last tag: steve-3.0.2

JakobRb, 2018-06-27:
Have you tried sending the same message twice? Literally, the same message.
These "some circumstances" are unknown. . . I guess the chargebox goes crazy for a moment :(

goekay, 2018-06-27:
yes, [as you can see in the code](https://github.com/RWTH-i5-IDSG/steve/blob/9541c2af59e24344fdd76b6adb2bd77ca035c8ad/src/test/java/de/rwth/idsg/steve/issues/Issue72.java#L79-L83) all the parameters of the StopTransactionRequest are the same. the same message is sent from 50 threads and 50 times per thread, concurrently. there are no exceptions.

the other thing that i noticed from your log excerpt is that cxf (the soap framework that we use) assigns the same ExchangeId to both messages. this is... crazy. i see such a thing for the first time. each message should get a new ExchangeId. is it a copy/paste error?

JakobRb, 2018-06-27:
It isn't a copy/paste error. This is why I said "literally the same message".
Maybe the chargebox don't sent the message twice and is a problem inside SteVe?

goekay, 2018-06-27:
added _low level_ stress test, which sends literally the same string, still no exceptions. i also could not see that cxf assigns the same ExchangeId to multiple messages.

JakobRb, 2018-06-27:
I will continue investigating on my own. I will report what I find.
Thank you very much for your help!

JakobRb, 2018-07-26:
I have been observing the logs for a while.
I have discovered a pattern: a "MeterValues" is received at the same time as the "StopTransaction".
I think that the second appearance of the "StopTransaction" is only because it accompanies the error log, but I have not been able to find it in the code.
```
[INFO ] 2018-07-25 15:35:58,827 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: 7c964ddc-68ff-4898-b7d1-3861d5412834
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">xxxxxxxxxxxxxxxx</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:be0f7418-9913-44f1-acf6-36d2aa878ca9</wsa5:MessageID><wsa5:From><wsa5:Address>https://xxx.xxx.xxx.xxx</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://xxxxxxxxxx.com:9999/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/MeterValues</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:meterValuesRequest><cs:connectorId>2</cs:connectorId><cs:transactionId>9999</cs:transactionId><cs:values><cs:timestamp>2018-07-25T15:40:17Z</cs:timestamp><cs:value unit="Wh" measurand="Energy.Active.Import.Register">9999999</cs:value></cs:values></cs:meterValuesRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[INFO ] 2018-07-25 15:35:58,827 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: 2c2fbd0c-8df8-4112-86f2-7fc541826926
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">xxxxxxxxxxxxxxxx</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:3d6bac67-f0b6-4771-8440-2f2dc0c7536c</wsa5:MessageID><wsa5:From><wsa5:Address>https://xxx.xxx.xxx.xxx</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://xxxxxxxxxx.com:9999/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StopTransaction</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:stopTransactionRequest><cs:transactionId>9999</cs:transactionId><cs:idTag>XXXXXXXX</cs:idTag><cs:timestamp>2018-07-25T15:40:17Z</cs:timestamp><cs:meterStop>9999</cs:meterStop></cs:stopTransactionRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[INFO ] 2018-07-25 15:35:58,838 org.apache.cxf.services.CentralSystemService.RESP_OUT - 
    ExchangeId: 7c964ddc-68ff-4898-b7d1-3861d5412834
    Payload: <soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope"><soap:Header><Action xmlns="http://www.w3.org/2005/08/addressing">/MeterValuesResponse</Action><MessageID xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:d9a83eeb-e6fc-44f2-8955-278dea370f5f</MessageID><To xmlns="http://www.w3.org/2005/08/addressing">http://www.w3.org/2005/08/addressing/anonymous</To><RelatesTo xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:be0f7418-9913-44f1-acf6-36d2aa878ca9</RelatesTo></soap:Header><soap:Body><meterValuesResponse xmlns="urn://Ocpp/Cs/2012/06/"/></soap:Body></soap:Envelope>
[INFO ] 2018-07-25 15:35:58,838 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: 2c2fbd0c-8df8-4112-86f2-7fc541826926
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">xxxxxxxxxxxxxxxx</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:3d6bac67-f0b6-4771-8440-2f2dc0c7536c</wsa5:MessageID><wsa5:From><wsa5:Address>https://xxx.xxx.xxx.xxx</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://xxxxxxxxxx.com:9999/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StopTransaction</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:stopTransactionRequest><cs:transactionId>9999</cs:transactionId><cs:idTag>XXXXXXXX</cs:idTag><cs:timestamp>2018-07-25T15:40:17Z</cs:timestamp><cs:meterStop>9999</cs:meterStop></cs:stopTransactionRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[WARN ] 2018-07-25 15:35:58,839 org.apache.cxf.phase.PhaseInterceptorChain - Application {urn://Ocpp/Cs/2012/06/}CentralSystemService#{urn://Ocpp/Cs/2012/06/}StopTransaction has thrown exception, unwinding now
org.apache.cxf.interceptor.Fault: SQL [update `transaction` set `transaction`.`stop_timestamp` = ?, `transaction`.`stop_value` = ?, `transaction`.`stop_reason` = ? where `transaction`.`transaction_pk` = ?]; Deadlock found when trying to get lock; try restarting transaction
        at org.apache.cxf.service.invoker.AbstractInvoker.createFault(AbstractInvoker.java:162) ~[steve-3.0.2.jar:?]
        at org.apache.cxf.jaxws.AbstractJAXWSMethodInvoker.createFault(AbstractJAXWSMethodInvoker.java:267) ~[steve-3.0.2.jar:?]

```

goekay, 2018-07-26:
hmm.. "MeterValues" does not write to transaction table. the only connection is that it references the primary key of the transaction table, if transaction id is set (which is your case). but i cannot imagine this causes the problem. my commit 698e9b6317717ca1daf72c309e78eb1f739a4c36 sends MeterValues and i still cannot reproduce the exception.

i will push a commit in a new git branch, where i will synchronize every method call in the class, such that java forces method executions to be sequential rather than parallel. please test with it to see whether the problem still occurs.

goekay, 2019-02-27:
closing the issue due to its inactivity.

closed 2019-02-27 by goekay