steve-community/steve #73: authorize in OCPP 1.5S does not return ConcurrentTx AuthorizationStatus

lnjuanj, 2018-07-02:
Hi all,
When an ocpp tag is currently in a transaction (steve backoffice: In Transaction? -> True):
- Our CB calls steve CS for an authorization for that card. (i.e. authorizeRequest)
- steve returns **Accepted** instead of **ConcurrentTx**.
Is this a possible bug in steve?
Thanks in advance for your help!

goekay, 2018-07-02:
hello, no it's not a bug, but was a conscious decision. the commit message of 7a6aaf9723b6136a5b750f274b1a9b43aecb51d2 explains this a bit. basically, we had a misbehaving station and decided to deactivate it.

goekay, 2018-07-02:
i have a more general solution instead of our previous workaround in 7a6aaf9723b6136a5b750f274b1a9b43aecb51d2. if the idtag X is in transaction at station Y
- return `Accepted` if `authorizeRequest` comes from Y.
- return `ConcurrentTx` if `authorizeRequest` comes from some other station than Y.

does this logic resolve your issue?

lnjuanj, 2018-07-02:
Hi @goekay I see. I just wanted to see whether this was a bug.
The logic you explained sounds very good, although it is more than the OCPP standard explicitly defined.
In case you implement it, I will use it for sure :+1:

ferhigher, 2018-07-02:
Hi @goekay, hi all !!
As a teammate of lnjuanj we observe strange behavior when doing a send_Local_list from CS to CB.
We find that ocppTags in Transaction have NULL in fields parentIdTag and ExpiryDate when they arrive to CB, even though in CS there is relative information for these fields.

Is this the expected behavior? Is this a bug?

Thanks!

goekay, 2018-07-02:
hi @ferhigher, current impl sets these fields only when the status is `Accepted`. this is not a bug. as you can see [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/OcppTagServiceImpl.java#L120-L122), the idea was to give as little information as possible for status values other than `Accepted`.

i will change that behaviour with my commit that changes the `ConcurrentTx` behaviour.

commit 830f66e4 (2018-07-02): fixes #73
closed 2018-07-02 by goekay