steve-community/steve #75: Relay OCPP message to another server

Fyod, 2018-07-04:
Would it be possible to add a function for relaying (repeating) the incoming message to another server? All it would need is a checkbox for enabling the function and a field for the second server.

Cheers!

goekay, 2018-07-04:
what is the use case of this feature?

Fyod, 2018-07-04:
Mainly for testing charger brands against a known working server, for troubleshooting in case messages have trouble being read.
Some chargers, ie. Keba, Efacec don't fully adhere to the OCPP standard and have very minor differences. I could then report whether these (and other) brands work OK.

goekay, 2018-07-04:
hm.. even though, i think that it might be useful for some users, i doubt everyone will benefit from it.
moreover, in order to cover all cases, we would have to configure different paths for different ocpp versions and distinguish between soap and json/websocket versions. soap case is easy, since it's just a http post. but for json/websocket case, we have to implement websocket connection and lifecycle management and all that stuff, since steve will act as man-in-the-middle between the station and vendor backend. 

i do not think that the benefits will outweigh the costs (of implementing it).

csamsel, 2018-07-04:
Just capture the communication using wireshark or tcpdump for alter inspection.

Fyod, 2018-07-06:
Thanks, guys. I'll update if there are any problems with certain chargeboxes.

closed 2018-07-07 by goekay