steve-community/steve #76: Steve API

victormunoz, 2018-07-06:
Is it possible to make external calls to steve?

For example, I would like to change the availability of a chargebox programatically, through a call to steve.

I have tried to reproduce with curl the call that steve makes after clicking the button "Perform" in the operations/changeAvailability section, but haven't succeded...

goekay, 2018-07-06:
unfortunetly, there is no api to do what you want. sure you can experiment with curl to mimic the http post that the html page makes, but do not forget that the web page user has to authenticate/login first to access the admin operations. so you have to mimic that, too.

goRaspy, 2018-07-07:
To play with curl it is useful to deactivate CSRF login token.

add : 

@Override
  protected void configure(HttpSecurity http) throws Exception {
    http
      .csrf().disable();
 in 
/src/main/java/de/rwth/idsg/steve/config/SecurityConfiguration.java

But take in consideration the security issue created by CSRF desactivation.

closed 2018-07-17 by goekay