steve-community/steve #79: Incorrect connector status

JakobRb, 2018-07-17:
Hello,
I've found a problem with the following lines:
https://github.com/RWTH-i5-IDSG/steve/blob/2a33aac1f631d4587be045e0cb467001d2a5c7b3/src/main/java/de/rwth/idsg/steve/repository/dto/TransactionStatusUpdate.java#L21
https://github.com/RWTH-i5-IDSG/steve/blob/2a33aac1f631d4587be045e0cb467001d2a5c7b3/src/main/java/de/rwth/idsg/steve/repository/dto/UpdateTransactionParams.java#L20
https://github.com/RWTH-i5-IDSG/steve/blob/2a33aac1f631d4587be045e0cb467001d2a5c7b3/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java#L273

because sometimes we are receiving the closing session messages in this order:

```
[INFO ] 2018-07-16 18:22:37,961 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: af21ff4e-1ee9-455b-bd28-06f66881d36b
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">xxxxxxxxxxxxxxxx</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:bbf7300a-d49d-4388-8259-63bc8fa4ff3e</wsa5:MessageID><wsa5:From><wsa5:Address>http://xxx.xxx.xxx.xxx:xxxxx</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://xxxxxxxxxx/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StatusNotification</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:statusNotificationRequest><cs:connectorId>1</cs:connectorId><cs:status>Available</cs:status><cs:errorCode>NoError</cs:errorCode><cs:info>Connector enabled</cs:info><cs:timestamp>2018-07-16T18:11:30Z</cs:timestamp><cs:vendorId>CIRCONTROL</cs:vendorId><cs:vendorErrorCode></cs:vendorErrorCode></cs:statusNotificationRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[INFO ] 2018-07-16 18:22:37,966 org.apache.cxf.services.CentralSystemService.RESP_OUT - 
    ExchangeId: af21ff4e-1ee9-455b-bd28-06f66881d36b
    Payload: <soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope"><soap:Header><Action xmlns="http://www.w3.org/2005/08/addressing">/StatusNotificationResponse</Action><MessageID xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:83da1c3b-1461-4e5f-862b-ee62eed45954</MessageID><To xmlns="http://www.w3.org/2005/08/addressing">http://www.w3.org/2005/08/addressing/anonymous</To><RelatesTo xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:bbf7300a-d49d-4388-8259-63bc8fa4ff3e</RelatesTo></soap:Header><soap:Body><statusNotificationResponse xmlns="urn://Ocpp/Cs/2012/06/"/></soap:Body></soap:Envelope>
[INFO ] 2018-07-16 18:22:38,127 org.apache.cxf.services.CentralSystemService.REQ_IN - 
    ExchangeId: e263d247-cc4c-4ea5-8351-32e0e4e3f835
    Payload: <?xml version="1.0" encoding="UTF-8"?>
<SOAP-ENV:Envelope xmlns:SOAP-ENV="http://www.w3.org/2003/05/soap-envelope" xmlns:SOAP-ENC="http://www.w3.org/2003/05/soap-encoding" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns:cp="urn://Ocpp/Cp/2012/06/" xmlns:cs="urn://Ocpp/Cs/2012/06/" xmlns:wsa5="http://www.w3.org/2005/08/addressing"><SOAP-ENV:Header><cs:chargeBoxIdentity SOAP-ENV:mustUnderstand="true">xxxxxxxxxxxxxxxx</cs:chargeBoxIdentity><wsa5:MessageID>urn:uuid:b4a8fc0b-6d36-4f8d1-89ab-f0959337b1bd</wsa5:MessageID><wsa5:From><wsa5:Address>http://xxx.xxx.xxx.xxx:xxxxx</wsa5:Address></wsa5:From><wsa5:To SOAP-ENV:mustUnderstand="true">https://xxxxxxxxxxxxxxxxxxxxxx/ocpp/services/CentralSystemService</wsa5:To><wsa5:Action SOAP-ENV:mustUnderstand="true">/StopTransaction</wsa5:Action></SOAP-ENV:Header><SOAP-ENV:Body><cs:stopTransactionRequest><cs:transactionId>9999</cs:transactionId><cs:idTag>XXXXXXXX</cs:idTag><cs:timestamp>2018-07-16T18:11:30Z</cs:timestamp><cs:meterStop>9999999</cs:meterStop></cs:stopTransactionRequest></SOAP-ENV:Body></SOAP-ENV:Envelope>

[INFO ] 2018-07-16 18:22:38,138 org.apache.cxf.services.CentralSystemService.RESP_OUT - 
    ExchangeId: e263d247-cc4c-4ea5-8351-32e0e4e3f835
    Payload: <soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope"><soap:Header><Action xmlns="http://www.w3.org/2005/08/addressing">/StopTransactionResponse</Action><MessageID xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:d18e92cd-8c41-4c57-9231-65011f45e0f1</MessageID><To xmlns="http://www.w3.org/2005/08/addressing">http://www.w3.org/2005/08/addressing/anonymous</To><RelatesTo xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:b4a8fc0b-6d36-4f8d1-89ab-f0959337b1bd</RelatesTo></soap:Header><soap:Body><stopTransactionResponse xmlns="urn://Ocpp/Cs/2012/06/"><idTagInfo><status>Accepted</status><expiryDate>2018-07-16T19:22:38.136Z</expiryDate></idTagInfo></stopTransactionResponse></soap:Body></soap:Envelope>
```
As you can see, the first message sets the connector status as `Available`, but the second message closes the transaction, and the status is set to `Finishing`, leaving an incorrect finals status value for the connector.

goekay, 2018-07-17:
i can understand the problem. since the timestamps of both events (statusNotificationRequest and stopTransactionRequest) are the same, the latter in the processing pipeline overrides the previous one. maybe we should remove this functionality (i.e. stop setting connector status values based on start/stop transaction messages)..

did anyone found it ever useful? the reason why we do it is explained [here](https://github.com/RWTH-i5-IDSG/steve/blob/2a33aac1f631d4587be045e0cb467001d2a5c7b3/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java#L280-L288)

JakobRb, 2018-07-17:
It's a deadlock :smile:

goekay, 2018-07-17:
so... instead of disabling this feature globally, i made it opt-out for each station. @JakobRb does it resolve your issue?

JakobRb, 2018-07-17:
Thanks for the quick implementation.
It's a great solution but, how many stations are NOT sending the `Available` status?
Can be a better solution set `FALSE` as default?

JakobRb, 2018-07-17:
It's a perfect solution. Thanks again.

closed 2018-07-17 by goekay