steve-community/steve #80: Sex not required

TechOverflow, 2018-07-18:
I just noticed that this user field is required and I think it may be better as optional. Is there a file where this can easily be changed?

goekay, 2018-07-27:
no. since this field consists of a drop down menu, you have to select something. i could add something like `-- Empty --` to this drop down list to express that it is optional as it the case in some other drop down menus. but then again, the drop down menu already contains `Other` which could be used to achive the same. i fail to see that this really is a problem.

goekay, 2018-08-05:
we already print the endpoint address of a station in its charge point details page.

TechOverflow, 2018-08-05:
I mean the server endpoints, which are shown in terminal when the app is started.

goekay, 2018-08-05:
okay, we can display them on the about page of steve. i will look into it.

goekay, 2018-08-06:
> Is it possible to echo out the endpoints in the web interface?

this is implemented in commit 27323aa484893495c273295027eef745dc2648e2. i am closing this issue, since i consider it resolved.

closed 2018-08-06 by goekay