steve-community/steve #81: Do not store duplicate charging sessions and connector status

goRaspy, 2018-07-19:
In certains circonstances, charging station will send OCPP message twice.

For example, with a GPRS data connection, we can have a best connectivity for Cell to chargepoint way and a less good connectivity for the chargepoint to cell canal.

So in this case, it may happen that the chargepoint will send ( for exemple ) : 
CP -> STEVE  ( good connectivity ) : Start transaction
Steve-> CP ( bad connectivity ) : accepted

But the CP never received the acknowledge by Steve and try to send it again ( sometimes many times ).

It should be useful to not store "duplicate" transactions in database. It is the same for connector status.

goekay, 2018-07-19:
thanks for the report. let's address the connector status issue first. what's the problem with it? the database table is constructed in a way that duplicate status values should not be a problem. do you see multiple identical entries on the web ui for one connector?

goRaspy, 2018-07-19:
Hi Goekay, many thanks for your (very) fast answer. 
Unfortunately I do not speak about GUI ( but you are right sometimes we can see double connectors status.

But my problem is about SQL database "pollution" especially for duplicates transactions whose need to be "cleaned" before used for statistic purpose.

Cheers

goRaspy, 2018-07-22:
THanks you for working on connector status.
Concerning transactions, there any possibilities to "do not write" in database "clones" (same data, different transaction PK) ? 

My need is to have a reliable view of chargepoint statistics without post treatment of data

goekay, 2018-07-22:
my referenced commit 9e836fcc791949a780694b04afef0f93fd068973 does exactly that. it looks at the database for a transaction with identical information as in the message (chargebox, connector, idtag, timestamp, start value) and writes into database only if it is a new one. otherwise it returns the transaction pk of the existing database row.

goRaspy, 2018-07-22:
My Bad :(

I did not understood that. I'll try your commit.

Many thanks

goekay, 2018-07-25:
hey @goRaspy, did you test my commits? does it solve your problem? if the answer is yes, i want to close this issue.

closed 2018-07-25 by goekay