steve-community/steve #82: Feature req.: User password

TechOverflow, 2018-08-15:
I  think it would be useful for various future implementations to have a simple (optional) md5 password field for users.

goekay, 2018-08-15:
can you please elaborate on this?

TechOverflow, 2018-08-15:
For example I plan on making some variation of an API and may need users to authenticate in the external app. While it would be possible to do the user authentication/login externally, I think it would be duplicating the basic user info is already implemented in steve, which is only missing a password field. This authentication would not allow users to log into the manager of course, just a simple user/password check against the current database.

TechOverflow, 2018-08-15:
On the topic of users, I also think it would be great to have steve-user roles in steve, although this is a more complicated request. User roles (rights) could for example allow: 
Admin to view/edit everything and do operations as is the current state
Manager to view everything, not edit, do OCPP operations
Operator to only do OCPP operations and view tasks