steve-community/steve #84: Client certificate

JakobRb, 2018-08-24:
Hello,
We are trying to connect with a chargepoint which requires a client certificate.
The problem is, anything we try, ends with this error:
`org.apache.cxf.transport.http.HTTPException: HTTP response '403: Forbidden' when communicating with https://testportalocpp.abbext.com/chargepointservice/`
Anyone with experience in this scenario?
Thanks in advance.

TechOverflow, 2018-08-24:
I don't know how/if steve handles certificates, but you need to include the certificates in the message options, ie. `CURLOPT_SSLCERT, "something.pem");` `CURLOPT_CAINFO, "something.cer");`.

goekay, 2018-08-24:
steve can handle https (and therefore certificates) if you set it that way. you should enable https and set keystore parameters [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/resources/config/prod/main.properties#L32-L35). keystore is the tool in java to manage keys and certificates ([more info](https://github.com/RWTH-i5-IDSG/steve/wiki/Configuration#https)).

csamsel, 2018-08-25:
So you mean the charge point expects a client certificate provided by the backend for the backwards direction (backend -> charge point)? that's indeed not possible at the moment. (and also not very meaningful tbh)

goekay, 2018-08-25:
@JakobRb what you are trying to do is indeed possible, but you have to configure the cxf client in a way, that it is aware of the keystore (not the case out of the box). we did it for the hubject integration some months ago. i cannot make the steve repo with the hubject integration public (since they did not ok it), but i can prepare a minimal working example. i will get back to you today.

JakobRb, 2018-08-25:
@goekay thanks a lot!!!!!

goekay, 2018-08-25:
so... instead of making a minimal working example, i extended the impl of ClientProvider to setup ssl, if the keystore path and password are set, since the general public would benefit from this, too.

in short: if you build the keystore with all certs you need and set the parameters in the properties file, all cxf clients will be ssl aware.

csamsel, 2018-08-25:
@JakobRb Would you need different client certificates for different charging stations?

JakobRb, 2018-08-25:
@goekay we will try tour commit. Thanks!
@csamsel yes, we need different client certificates (not for each charging station, but for different groups)

JakobRb, 2018-08-25:
Now we are getting a new error (summarized):
```
org.apache.cxf.interceptor.Fault: Could not send Message.
Caused by: javax.net.ssl.SSLHandshakeException: SSLHandshakeException invoking https://testportalocpp.abbext.com/chargepointservice/: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Caused by: javax.net.ssl.SSLHandshakeException: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Caused by: sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
Caused by: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target
```

goekay, 2018-08-25:
i am not familiar with this exception. googling it returns multiple results with diagnosis and possible solutions, i would suggest to look at them.

JakobRb, 2018-08-25:
Now it's working!!! Thanks a lot.
The last error was caused because I was not aware that I was not using the cacerts from Java, but the keystore that I configured. I added the root certificate from the server that I want to connect, and everything worked properly.
Thanks again!

csamsel, 2018-08-26:
Good to hear. @goekay maybe we sould keep a TODO to make client certifacts configurable / assignable per charging station.

JakobRb, 2018-08-26:
AFAIK, the client certificate sent is selected based in the server certificate. The client sends the certificate which is signed by the certificate received from the server.

goekay, 2018-08-26:
correct. afaik, you just should put all certs in the keystore, the correct cert will be chosen during the tls handshake.

closed 2018-08-26 by goekay