steve-community/steve #86: Incorrect datetime value

v92, 2018-09-07:
I'm using EVSE from http://www.ebee.berlin/ with CC612 controller (FW 4.31) and I've received this error. SteVe has received timestamp in 1970-01-01T00:00:31Z format and for some reason timestamp datatype couldn't handle it. 

As a workaround I've changed column data type to datetime with:

`alter table connector_status modify status_timestamp datetime;`

but I'm not sure if it is right solution.

```
[INFO ] 2018-09-07 17:41:59,970 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=421917391005, sessionId=599decd5] Connection is established
[INFO ] 2018-09-07 17:42:00,711 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=421917391005, sessionId=599decd5] Received: [2, "77314252", "StatusNotification", {"connectorId": 1, "status": "Available", "errorCode": "NoError", "info": "Status Update", "timestamp": "1970-01-01T00:00:31Z", "vendorId": "Bender GmbH Co. KG"}]
[INFO ] 2018-09-07 17:42:00,722 de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl - The connector 421917391005/1 is NEW, and inserted into DB.
[ERROR] 2018-09-07 17:42:00,737 de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler - Exception occurred
org.jooq.exception.DataAccessException: SQL [insert into `stevedb`.`connector_status` (`connector_pk`, `status_timestamp`, `status`, `error_code`, `error_info`, `vendor_id`, `vendor_error_code`) values ((select `stevedb`.`connector`.`connector_pk` from `stevedb`.`connector` where (`stevedb`.`connector`.`charge_box_id` = ? and `stevedb`.`connector`.`connector_id` = ?)), ?, ?, ?, ?, ?, ?)]; Data truncation: Incorrect datetime value: '1970-01-01 00:00:31' for column 'status_timestamp' at row 1
	at org.jooq_3.11.4.MYSQL.debug(Unknown Source) ~[?:?]
	at org.jooq.impl.Tools.translate(Tools.java:2384) ~[steve.jar:?]
	at org.jooq.impl.DefaultExecuteContext.sqlException(DefaultExecuteContext.java:811) ~[steve.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:364) ~[steve.jar:?]
	at org.jooq.impl.AbstractDelegatingQuery.execute(AbstractDelegatingQuery.java:127) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.lambda$insertConnectorStatus$0(OcppServerRepositoryImpl.java:129) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext$3.run(DefaultDSLContext.java:584) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext$3.run(DefaultDSLContext.java:581) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext.lambda$transactionResult0$0(DefaultDSLContext.java:521) ~[steve.jar:?]
	at org.jooq.impl.Tools$10$1.block(Tools.java:4355) ~[steve.jar:?]
	at java.util.concurrent.ForkJoinPool.managedBlock(ForkJoinPool.java:3323) ~[?:1.8.0_181]
	at org.jooq.impl.Tools$10.get(Tools.java:4352) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext.transactionResult0(DefaultDSLContext.java:564) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext.transactionResult(DefaultDSLContext.java:491) ~[steve.jar:?]
	at org.jooq.impl.DefaultDSLContext.transaction(DefaultDSLContext.java:581) ~[steve.jar:?]
	at de.rwth.idsg.steve.repository.impl.OcppServerRepositoryImpl.insertConnectorStatus(OcppServerRepositoryImpl.java:107) ~[steve.jar:?]
	at de.rwth.idsg.steve.service.CentralSystemService16_Service.statusNotification(CentralSystemService16_Service.java:115) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.soap.CentralSystemService16_SoapServer.statusNotification(CentralSystemService16_SoapServer.java:78) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.ocpp16.Ocpp16WebSocketEndpoint$Ocpp16CallHandler.dispatch(Ocpp16WebSocketEndpoint.java:67) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.AbstractCallHandler.accept(AbstractCallHandler.java:28) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.pipeline.IncomingPipeline.accept(IncomingPipeline.java:43) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleTextMessage(AbstractWebSocketEndpoint.java:90) ~[steve.jar:?]
	at de.rwth.idsg.steve.ocpp.ws.AbstractWebSocketEndpoint.handleMessage(AbstractWebSocketEndpoint.java:62) ~[steve.jar:?]
	at org.springframework.web.socket.adapter.jetty.JettyWebSocketHandlerAdapter.onWebSocketText(JettyWebSocketHandlerAdapter.java:83) ~[steve.jar:?]
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:1.8.0_181]
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:1.8.0_181]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_181]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_181]
	at org.eclipse.jetty.websocket.common.events.annotated.CallableMethod.call(CallableMethod.java:71) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.annotated.OptionalSessionCallableMethod.call(OptionalSessionCallableMethod.java:72) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextMessage(JettyAnnotatedEventDriver.java:234) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.message.SimpleTextMessage.messageComplete(SimpleTextMessage.java:69) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.appendMessage(AbstractEventDriver.java:66) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.JettyAnnotatedEventDriver.onTextFrame(JettyAnnotatedEventDriver.java:226) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.events.AbstractEventDriver.incomingFrame(AbstractEventDriver.java:162) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.WebSocketSession.incomingFrame(WebSocketSession.java:476) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.extensions.ExtensionStack.incomingFrame(ExtensionStack.java:220) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.Parser.notifyFrame(Parser.java:220) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.Parser.parse(Parser.java:245) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.readParse(AbstractWebSocketConnection.java:560) ~[steve.jar:?]
	at org.eclipse.jetty.websocket.common.io.AbstractWebSocketConnection.onFillable(AbstractWebSocketConnection.java:391) ~[steve.jar:?]
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:281) ~[steve.jar:?]
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:102) ~[steve.jar:?]
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:118) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:333) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:310) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:168) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:126) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.ReservedThreadExecutor$ReservedThread.run(ReservedThreadExecutor.java:366) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:762) ~[steve.jar:?]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:680) ~[steve.jar:?]
	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_181]
Caused by: com.mysql.cj.jdbc.exceptions.MysqlDataTruncation: Data truncation: Incorrect datetime value: '1970-01-01 00:00:31' for column 'status_timestamp' at row 1
	at com.mysql.cj.jdbc.exceptions.SQLExceptionsMapping.translateException(SQLExceptionsMapping.java:104) ~[steve.jar:?]
	at com.mysql.cj.jdbc.ServerPreparedStatement.serverExecute(ServerPreparedStatement.java:633) ~[steve.jar:?]
	at com.mysql.cj.jdbc.ServerPreparedStatement.executeInternal(ServerPreparedStatement.java:414) ~[steve.jar:?]
	at com.mysql.cj.jdbc.ClientPreparedStatement.execute(ClientPreparedStatement.java:392) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.ProxyPreparedStatement.execute(ProxyPreparedStatement.java:44) ~[steve.jar:?]
	at com.zaxxer.hikari.pool.HikariProxyPreparedStatement.execute(HikariProxyPreparedStatement.java) ~[steve.jar:?]
	at org.jooq.tools.jdbc.DefaultPreparedStatement.execute(DefaultPreparedStatement.java:209) ~[steve.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:432) ~[steve.jar:?]
	at org.jooq.impl.AbstractDMLQuery.execute(AbstractDMLQuery.java:613) ~[steve.jar:?]
	at org.jooq.impl.AbstractQuery.execute(AbstractQuery.java:350) ~[steve.jar:?]
	... 48 more
[INFO ] 2018-09-07 17:42:00,743 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=421917391005, sessionId=599decd5] Sending: [4,"77314252","InternalError","Internal services failed while processing of the payload",{"errorMsg":"SQL [insert into `stevedb`.`connector_status` (`connector_pk`, `status_timestamp`, `status`, `error_..."}]
[WARN ] 2018-09-07 17:43:01,947 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=421917391005, sessionId=599decd5] Connection is closed, status: CloseStatus[code=1006, reason=WebSocket Read EOF]
```

goekay, 2018-09-07:
please make sure that the [time zone of steve](https://github.com/RWTH-i5-IDSG/steve/blob/dec80f384e1eceb57ffac8d6f3937c6cb39cefc2/src/main/java/de/rwth/idsg/steve/SteveConfiguration.java#L28) (default: utc) is the same as the time zone of your mysql server. i think your mysql runs in your local time zone and therefore it tries to convert `1970-01-01 00:00:31.0` to utc, which results in a timestamp earlier than `1970-01-01 00:00:01`, which causes the exception. running java and mysql in different timezones is very tricky and error-prone due to conversions that need to happen. that's why we suggest using the same time zone for both and preferably utc. utc is the suggested time zone by ocpp anyway.

v92, 2018-09-08:
Thank you, I've set timezone to UTC and it seems to be working.

closed 2018-09-08 by v92

eyalmanor, 2021-10-10:
thank you for the quick reply

I have already gone through all the documents related to the topic including the link you presented but it does not solve the problem I am attaching the Eclipse log.
Exception in thread "main" java.lang.Error: Unresolved compilation problems: 
	log cannot be resolved
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	log cannot be resolved
	The method getTimeZoneId() is undefined for the type SteveConfiguration
	Cannot switch on a value of type Object. Only convertible int values, strings or enum variables are permitted
	DEV cannot be resolved to a variable
	TEST cannot be resolved to a variable
	PROD cannot be resolved to a variable

	at de.rwth.idsg.steve.Application.<init>(Application.java:41)
	at de.rwth.idsg.steve.Application.main(Application.java:63)

goekay, 2021-10-11:
> thank you for the quick reply
>
> I have already gone through all the documents related to the topic including the link you presented but it does not solve the problem I am attaching the Eclipse log. [...]

this has nothing to do with the discussed topic. steve uses a tool (lombok) to generate boilerplate code at compile time. eclipse does not understand this and therefore gets grumpy. please build the project on terminal using commands that we listed on readme.