steve-community/steve #89: Adding custom pages

TechOverflow, 2018-09-21:
I'm trying to add custom folders/pages to the web view, but my linking isn't working.
In the (src) `/views` folder, I have added `/test/subfolder/test.jsp`
Then I added <`a href="${ctxPath}/manager/test/subfolder/test">Click</a>` to the menu.
I get a 404, even though I can see in the (target) `WEB-INF/web.xml` that the servlet routes are created upon build.

Is there something else that needs to be done?

goekay, 2018-09-22:
> Is there something else that needs to be done?

yes, jsp files are only 1/3 of the story. the web part of steve follows model-view-controller (MVC) pattern and jsps are only the views (templates). they cannot be accessed directly from http. there has to be a controller class with the url mappnig that listens to, for example, `manager/test/subfolder/test` that executues some logic, builds a model (container of the data which you want to render on a web page) and sends this model to the jsp file. the string method return values you see in controller classes (like [this](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/controller/ChargePointsController.java#L67)) are the locations of the view templates, which spring framework uses to render the model.

TechOverflow, 2018-09-24:
Got it. Thank you!

closed 2018-09-24 by goekay