steve-community/steve #90: "Log file: Not available" with JDK11

sergtop, 2018-10-18:
Recent version with JDK11 can`t start logging due to 

WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.

switching back to 1.8 in pom.xml solves issue

goekay, 2018-10-18:
can reproduce the warnings. but the log file is there. it does start logging. are you sure about this?

sergtop, 2018-10-18:
yes, the file is created, but it can`t be accessed via */manager/log URI from webservice

goekay, 2018-10-18:
oh okay, thanks, i will look into it.

goekay, 2018-10-18:
i want to mention, that the warning is still there. but reading the configuration where to log file is is now successful. 

there are some other warnings related to jooq as well. these should be fixed by the library maintainers with future releases.

commit e4321ac0 (2018-10-18): use other log4j api to read logging configuration (closes #90)