steve-community/steve #95: Enable multiple (concurrent) transactions

chuck-h, 2018-12-04:
_feature request_
The existing behavior for blocking concurrent transactions was discussed at #73. However it overlooks those use cases (e.g. fleet vehicles) where the business logic is to _allow_ a single user to authorize charging on several vehicles concurrently. OCPP1.6 docs specify the meaning of `AuthorizationStatus.ConcurrentTx` to be **Identifier is already involved in another transaction and multiple transactions are not allowed.** However, if multiple transactions _are_ allowed, then `ConcurrentTx` status should not be used (`Accepted` is appropriate if the Id is otherwise ok).

I propose that "allow multiple transactions" should be a DB field in ocpp_tag. The UI should allow configuring -- e.g. thru Settings page -- the default value of this field used when a new tag is entered. The tag management pages should allow viewing and setting the value on individual tags.

The value in this field would affect logic in the `OcppTagRepository` and `OcppTagService` implementations.

goekay, 2018-12-04:
just throwing this idea out there: do we need `ConcurrentTx` at all? as i stated in the [other issue](https://github.com/RWTH-i5-IDSG/steve/issues/73#issuecomment-401803575), i am not sure how i feel about it.

chuck-h, 2018-12-04:
After thinking about this a bit more I would say:

1. I have no use for `ConcurrentTx` in my immediate use case.
1. I find the existence of SteVe's "no concurrent charging" logic annoying for my use case, but I can work around it if necessary. (Basically, some users will have to carry two cards.)
1. There _is_ a need, in general, to implement "this transaction is blocked by a back-end business rule". "No concurrent charging" is one possible business rule -- arguably not a particularly useful one -- but it does not deserve special treatment in the protocol.`Blocked` is a perfectly satisfactory way of communicating the situation to the CP.
1. I would not object at all to @goekay saying "SteVe does not implement specific business rules" and ripping out all the concurrent charging code.
1. I would support a refactoring that makes it relatively easy and safe for a user to plug in custom authorization business logic which manages `Blocked` status. I don't know quite what that would look like.

goekay, 2018-12-04:
ok. since i cannot decide this, i want to give the deployer/user of steve the possibility to decide. i want to introduce a new numeric database field `max_active_tx_count` or something similar, which can be used to express one of the multiple things:

- if set to 0, it can have the same effect as `Blocked`,  practically saying "this tag cannot have any active transactions"
- if set to -1 (or some negative value) it will disable concurrent tx check, in this case the tag will always be accepted
- if set to some positive value, its value will be used how many active concurrent tx the tag can have.

this solution is a little bit more expressive than having a boolean flag, since the user can decide the concurrency level. and the user can also decide to disable this concurrency check if he wishes. how about that?

while i implement this, i also want to retire the `blocked` and `is_transaction` fields.

goekay, 2018-12-05:
> BTW any comments on the broader idea of a clean hook for per-transaction-authorization business logic? 

i sure can split the [method](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/service/OcppTagServiceImpl.java#L109-L136) that decides the status into more granular components, but people who wish to do so can also extend/change the method with their own logic. i think the method is as clean and simple as it gets. what do you have in mind? or if i ask another way: what do you want to improve?

chuck-h, 2018-12-06:
Just noticed #23, seems relevant.

goekay, 2018-12-06:
a hypothetical extension mechanism based on some assumptions and speculation does not sound so useful. everyone's use cases and needs are different and it's hard to address all with one solution. you are coming from your use cases and experience with your charging stations, which is good but hard to generalise to come up with a solution everyone would benefit from. for example:

> I suppose the full parameter set of a StartTransaction.req message might be relevant.

the authorisation status is calculated not only for StartTransaction but also for Authorize and StopTransaction messages. so what is the common denominator context here? should we have three different methods with different contexts? not a fan of this idea.

> Maybe something about the WebSocketSession attributes (e.g. for a security model)

the transport mechanism (soap or websocket) is abstracted away at this point in the execution pipeline, because it is handled by the lower layers. this is a good thing, because it separates nitty-gritty of transportation details from business logic. this unfortunately is also the result of abstraction.