steve-community/steve #97: [Question] Get the power consumption by Tag (OCPP 1.5)

Ncifra, 2018-12-06:
Hi,

We have a charging point with OCPP 1.5. Is it possible to get power consumption statistics, by tag, transaction, or any kind of combination between these? I've been researching for something that may be related to this in the OCPP version specifications but nothing comes up. Also #92 and #44 seems to have mentions of the consumption (also specific to OCPP 1.6), but nothing concrete.

Thank you

goekay, 2018-12-06:
as far as i know, there is nothing related to power consumption statistics in ocpp spec. but ocpp already provides the raw usage data, which you can use to calculate these...

if you mean _browsing_ consumption statistics, the transaction page can be used to list transactions filtered by transaction id, tag, charging station etc.. then you can click on the transaction id of each row to see the details. on this detailed transaction info page, you see the start, intermediate and stop meter values (with their timestamps) that were sent to steve. since you have this data, calculating the actual consumption statistics only requires applying some arithmetic operations.

Ncifra, 2018-12-07:
Yes, you are correct. Didn't notice that the transaction numbers were clickable. Was clicking the ChargeBox ID and weren't noticing the discrepancies in the data provided. 

One other thing I noticed and didn't find anything about it: The "default" transaction that always shows as open is it an accumulator of sorts? Will it become a bit bloated / overloaded when reaching too many transaction samples?

closed 2018-12-07 by Ncifra

goekay, 2018-12-07:
> The "default" transaction that always shows as open...

we did not implement anything in this regard. the transaction table displays what the charging stations communicate us. therefore, i assume that one charging station is misbehaving and did not send a StopTransaction message.

Ncifra, 2018-12-07:
So this is the list of all transactions. Number "1", is the "default" I was mentioning:

[image]

Here is its view:

[image]

And here are the other ones, 2 and 3 that are closed. Notice from above the these two are included in the above view:

[image]
[image]

It seems to me that 1 is like a holder for all transactions. At first I was seeing the "zombie" transaction issue, but then seeing this, I remembered that I saw in one of the issues (can't remember which), that there was a reserved spot for transactions or something like that.

goekay, 2018-12-07:
> It seems to me that 1 is like a holder for all transactions. 

yes, it seems like it. especially seeing the same intermediate values from transaction 3 in transaction 1 with multiple `Transaction.Begin` contexts strengthens this assumption. but i do not know why the developers of the charging station decided to do it this way. this behaviour does not conform to ocpp spec.

Ncifra, 2018-12-07:
I understand, so it's them that are sending this "fake" transaction that gets interpreted correctly on the other hand by Steve, since it is most as a passive receiver of data. 

The multiple `Transaction.Begin`'s actually are what made me curious. I don't know if this will pose memory problems, even on the charging station itself, or during the transaction data exchange, if it goes beyond some limit.

goekay, 2018-12-07:
just to be sure: steve logs the request/response messages to the log file. if the assumption of "holder" transaction is true, you should see the message arriving twice (once for the actual transaction and one for the holder transaction). is this the case?

goekay, 2018-12-07:
or you can also look at the `connector_meter_value` database table and try to identify whether there are multiple entries.

goekay, 2018-12-07:
upon further inspection of the logic how we retrieve intermediate values from database, i am afraid to say that the case of zombie transactions is not handled correctly. therefore, meter values for all subsequent transactions at the same chargebox and connector are falsely attributed to the zombie transaction. man, what an edge case. thanks for reporting it.

Ncifra, 2018-12-07:
Just checked the `connector_meter_value` table and it didn't have the number 1 transaction's data.

While the log itself doesn't seem to have anything in that regard:

```
[INFO ] 2018-12-07 09:03:11,053 org.apache.cxf.wsdl.service.factory.ReflectionServiceFactoryBean - Creating Service {urn://Ocpp/Cp/2012/06/}ChargePointServiceService from class ocpp.cp._2012._06.ChargePointService
[INFO ] 2018-12-07 09:03:11,096 org.apache.cxf.services.ChargePointService.REQ_OUT - 
    ExchangeId: 818643fd-1ad6-4299-8e7b-91b27821440f
    Payload: <soap:Envelope xmlns:soap="http://www.w3.org/2003/05/soap-envelope"><soap:Header><Action xmlns="http://www.w3.org/2005/08/addressing">/RemoteStartTransaction</Action><MessageID xmlns="http://www.w3.org/2005/08/addressing">urn:uuid:4f6acde9-0742-4ada-8188-6b0dab4cca73</MessageID><To xmlns="http://www.w3.org/2005/08/addressing">http://192.168.1.7:8765</To><ReplyTo xmlns="http://www.w3.org/2005/08/addressing"><Address>http://www.w3.org/2005/08/addressing/anonymous</Address></ReplyTo><chargeBoxIdentity xmlns="urn://Ocpp/Cp/2012/06/">CBI-3c030307f24b</chargeBoxIdentity></soap:Header><soap:Body><remoteStartTransactionRequest xmlns="urn://Ocpp/Cp/2012/06/"><idTag>70889D5A</idTag><connectorId>1</connectorId></remoteStartTransactionRequest></soap:Body></soap:Envelope>
```

There's only the log of a failed remote transaction that we tried today, but nothing for the older requests.

goekay, 2018-12-07:
> Just checked the connector_meter_value table and it didn't have the number 1 transaction's data.

there are multiple fallbacks in the logic. if the transaction id is not referenced in meter values messages, then we select meter values between transaction start and end. if there is no transaction end (i.e. active transaction), then we use the start timestamp and read all meter values _after this timestamp_. as you can imagine, a zombie transaction totally confuses the last point.

Ncifra, 2018-12-07:
I see. Is this maybe a logical shortcut you have used in the calculations, due to the fact that you take into account that there can't be more than one active transaction (I saw an issue regarding the ability to have concurrent transactions, so by default you should have "max" one transaction active), so in theory each metered data after the beginning of the transaction, should belong to this "active" transaction.

goekay, 2018-12-07:
the logic that i applied was "the active transaction should be the last and latest one at this charge box and connector". this thinking was not correct in case of zombie transactions.

i did not assume that there can be only one active transaction _per user._ this is independent from users having multiple concurrent transactions, because they will be charging at different chargebox and connectors (hopefully. or will they not? i am totally confused atm  :D).

> also the next one's sorted by transaction start date (if there is a later one, than the previous is a zombie)

this is the idea that i am implementing now. if the transaction did not end, i search for the start timestamp of the next transaction at the same chargebox and connector, and filter the meter values between `start of zombie transaction` and `start of next transaction after zombie`.

goekay, 2018-12-07:
@Ncifra is it possible that you try my referenced commit?

Ncifra, 2018-12-07:
> i did not assume that there can be only one active transaction per user. this is independent from users having multiple concurrent transactions, because they will be charging at different chargebox and connectors (hopefully. or will they not? i am totally confused atm :D).

Yes, indeed, different chargeboxes, but also from the same chargebox but different connections could create simultaneous transactions.

> if the transaction did not end, i search for the start timestamp of the next transaction at the same chargebox and connector, and filter the meter values between start of zombie transaction and start of next transaction after zombie

As I mentioned above shouldn't also the transaction id that is available in the metered connection values table be used too? Or in fact just that one? Or is there any case where it can be inconsistent.

> @Ncifra is it possible that you try my referenced commit?

Unfortunately I don't have direct access to the server code for now. But as I mentioned above, shouldn't finding/joining by transaction_pk on the metered connection values be more robust?

goekay, 2018-12-07:
`transaction_pk` is indeed [used](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java#L131-L136). this is the best-case.

but it is not guaranteed that the station sets the transaction id for meter value (this was our case). here we fallback to [timestamp-based filtering](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java#L138-L147).

Ncifra, 2018-12-07:
All right, I will try and gain access to this and do a quick regeneration with the above code, and let you know.

Ncifra, 2018-12-07:
Success!

[image]

Ncifra, 2018-12-07:
> basically, we apply multiple strategies to work around the differences of various station implementations.

These things though should have been enforced by the protocol.

Ncifra, 2018-12-10:
No no, I understand it from your side, that's why I said "enforcement by the protocol", so as not to have too much weight on the developer, and not branching out in conditions of this sort that increase the possibility of errors. For example OCPP 1.5 compliant is a software that for example always sends the transaction ID, something like that. These are small things but that lift a lot of weight from the development.

> why is the station not aware of a transaction it started and why does it not stop it?

Well this sort of things I think can happen when packages are dropped (`stop` request for example) , even though on TCP this shouldn't be the case.

TechOverflow, 2018-12-13:
In my experience, if you want a stable 1.5, use LAN if possible and preferably though a VPN to the server (may require additional hardware). The protocol isn't good at handling weird packet drops and things that arise on WiFi and especially GSM. That's probably one of the main reasons they went with websockets on 1.6, which still has quirks depending on manufacturer and various client settings. I have experienced 1.5 stations on WiFi and GSM that weren't able to communicate reliably enough when there were too many people either on the WiFi (yes, it should've been dedicated, but that wasn't possible) or the cell tower was very occupied (busy parts of a city).

Ncifra, 2018-12-14:
> I have experienced 1.5 stations on WiFi and GSM that weren't able to communicate reliably enough when there were too many people either on the WiFi

Yes, depending on the router that is to be expected.

Currently we have our station in LAN (and no VPN that I know of), so there should be a reliable connection. It remains to be seen if this thing will show up again.