steve-community/steve #99: MeterValues

lincomatic, 2018-12-17:


goekay, 2018-12-17:
there are multiple things that i need to mention:
- can you make sure that the entries are in the database? they should be written no matter what.
- reading of meter values contains a little bit of logic ([for example](https://github.com/RWTH-i5-IDSG/steve/blob/72e086e496c46babe13c8c0c01f0126010f7108f/src/main/java/de/rwth/idsg/steve/repository/impl/TransactionRepositoryImpl.java#L163-L169)). if there are multiple entries with the identical timestamp and values, only the first one is selected.

lincomatic, 2018-12-17:
There's only 1 timestamp in each payload. If you tell me where to look in the database, I will check.

Here are the matching entries from the log:

```
> [INFO ] 2018-12-14 18:09:11,080 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=2002, sessionId=6f6dab2b] Received: [2,"lIqURiSp2RJeWVccSIoL8Z5EDYkEsOiM","MeterValues",{"connectorId":1,"transactionId":296,"meterValue":[{"timestamp":"2018-12-15T02:09:10.345Z","sampledValue":[{"value":"2134","unit":"Wh","measurand":"Energy.Active.Import.Register","context":"Sample.Periodic"},{"value":"80","unit":"A","measurand":"Current.Offered","context":"Sample.Periodic"}]}]}]
> [INFO ] 2018-12-14 18:09:11,180 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=2002, sessionId=6f6dab2b] Sending: [3,"lIqURiSp2RJeWVccSIoL8Z5EDYkEsOiM",{}]
> [INFO ] 2018-12-14 18:09:22,075 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=2002, sessionId=6f6dab2b] Received: [2,"10avewBaSnSD37I80q34W7PX4PGuD6zN","MeterValues",{"connectorId":1,"transactionId":296,"meterValue":[{"timestamp":"2018-12-15T02:09:21.253Z","sampledValue":[{"value":"2143","unit":"Wh","measurand":"Energy.Active.Import.Register","context":"Sample.Periodic"},{"value":"80","unit":"A","measurand":"Current.Offered","context":"Sample.Periodic"}]}]}]
> [INFO ] 2018-12-14 18:09:22,220 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=2002, sessionId=6f6dab2b] Sending: [3,"10avewBaSnSD37I80q34W7PX4PGuD6zN",{}]
> [INFO ] 2018-12-14 18:09:32,870 de.rwth.idsg.steve.ocpp.ws.WebSocketLogger - [chargeBoxId=2002, sessionId=6f6dab2b] Received: [2,"9jlu8oksxqsDEV6gizTIngUa9iHNG1PH","MeterValues",{"connectorId":1,"transactionId":296,"meterValue":[{"timestamp":"2018-12-15T02:09:32.131Z","sampledValue":[{"value":"2151","unit":"Wh","measurand":"Energy.Active.Import.Register","context":"Sample.Periodic"},{"value":"80","unit":"A","measurand":"Current.Offered","context":"Sample.Periodic"}]}]}]
> 
```

goekay, 2018-12-18:
> Good way to save space in the DB

steve always inserts all meter values into the db without much logic/processing. so, you can find them in the db table `connector_meter_value`. we only show the first one on the web pages. this was somewhat a design decision: i do not want to interpret/process the messages before/while inserting into db, because the logic might contain errors or we might change it later. we just insert them into db as they come. _reading of values_ and displaying them on web pages includes some interpretation/processing, which still might contain errors, but this is no problem since there if no information loss and you always have the fallback of looking at the database tables if you are not satisfied with the web pages.

it works as designed, since you have 6 meter value measurements 3 of which are the same (Current.Offered) and therefore only the first is shown in the table, and remaining 3 are different (Sample.Periodic) and therefore all are shown in the table.

closed 2018-12-18 by goekay