steve-community/steve #100: OCPP 1.6-J Security

V2G-UK, 2018-12-19:
The OCA have just back ported OCPP 2.0 features to OCPP 1.6-J only. Do the SteVe team have any plans to support this new part of the standard? See:

http://www.v2g-evse.com/2018/12/18/open-charge-alliance-enhances-ocpp-1-6-security/

> Secure communication and operation is a critical aspect of Electric Vehicle Charging Infrastructure.
> 
> To further assist the industry the Open Charge Alliance now publishes a white paper to describe a standard way to address security using OCPP 1.6-J.

According to the new "white paper":

> This document is for OCPP 1.6-J (JSON over WebSockets) only, OCPP-S (SOAP) is NOT supported. This document was started, as it is seen as a simple stap to port OCPP 2.0 security to OCPP 1.6. But as OCPP 2.0 only support JSON over WebSockets (not SOAP), this document is also written for OCPP 1.6-J only. Adding SOAP to this document would have taken a lot of work and review by security experts.
> 
> This document is based on OCPP 2.0. To help developers that are implementing both 1.6J security improvement and OCPP 2.0, we have kept the Use Case numbering from OCPP 2.0. So when implementing for example Use Case N01, it is the same use case in this document as in the 2.0 specification.

lategoodbye, 2023-02-13:
Since this feature is requested regularly, maybe we should discuss some specific points here:

- What is the expected security profile?
- Is it possible to implement a conform Basic Auth in Steve (reject invalid credential before upgrading the websocket)? 
- Which messages must be implemented (e.g. handle SecurityEventNotification from the ChargePoint)?

andle, 2024-11-26:
@goekay Thanks for the great work as always! 

Just wanted to (poke) see if there were any updates on this?
I'm finding more CPO's requesting Basic Auth and wanted to be able to test my EVSE's implementation.

I like your idea about adding username/password configuration into the ChargePoint's page (can't find which issue discussed that).

nikhilkandari77, 2025-02-17:
> The OCA have just back ported OCPP 2.0 features to OCPP 1.6-J only. Do the SteVe team have any plans to support this new part of the standard? See: [...]

Hi,

I have successfully ported the OCPP 2.0 security updates into the OCPP 1.6J Steve implementation, as outlined in the OCPP 1.6J white paper.

goekay, 2025-02-17:
hey @nikhilkandari77 this is great. are you planning to create a PR?

MikeLeonFox, 2025-05-08:
Hey there, just looking into this. Is there a known timeframe when this will be implemented?