steve-community/steve #102: Build of 3.2.0 fails

jhs-s, 2018-12-28:
Clean tarball, just downloaded and changed credentials, same if I use the 3.2.0 git tag.

[ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.8.0:compile (default-compile) on project steve: Compilation failure: Compilation failure:
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[60,25] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[92,49] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[136,34] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[165,59] cannot find symbol
[ERROR] symbol:   method setInTransaction(boolean)
[ERROR] location: class jooq.steve.db.tables.records.OcppTagRecord
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[166,54] incompatible types: inference variable T has incompatible bounds
[ERROR] equality constraints: jooq.steve.db.tables.records.OcppTagRecord
[ERROR] lower bounds: java.lang.Object
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppTagRepositoryImpl.java:[180,36] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/web/controller/OcppTagsController.java:[92,51] cannot find symbol
[ERROR] symbol:   method getInTransaction()
[ERROR] location: variable record of type jooq.steve.db.tables.records.OcppTagRecord
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/service/OcppTagServiceImpl.java:[126,19] cannot find symbol
[ERROR] symbol:   method getInTransaction()
[ERROR] location: variable record of type jooq.steve.db.tables.records.OcppTagRecord
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/service/OcppTagServiceImpl.java:[143,26] cannot find symbol
[ERROR] symbol:   method getInTransaction()
[ERROR] location: variable record of type jooq.steve.db.tables.records.OcppTagRecord
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[201,37] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] /opt/steve-steve-3.2.0/src/main/java/de/rwth/idsg/steve/repository/impl/OcppServerRepositoryImpl.java:[257,50] cannot find symbol
[ERROR] symbol:   variable IN_TRANSACTION
[ERROR] location: variable OCPP_TAG of type jooq.steve.db.tables.OcppTag
[ERROR] -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoFailureException

goekay, 2018-12-29:
what is mysql/mariadb version you are using? did you apply database migrations? is this issue related to #101? 

i assume that the db migrations are not done successfully, because the java compilation errors indicate database fields not existing in tables, which is correct because they changed with the latest migration scripts.

jhs-s, 2018-12-29:
The database migrations was succesful (at least no errors were reported). I wasn't aware that compilation errors could be generated by a wrong database so I didn't retry after updating mariadb because I thought that would be unrelated.

Closing...thanks for your reply!

closed 2018-12-29 by jhs-s