steve-community/steve #103: app integration via restful api

Gior80, 2019-01-09:
Hi guys, 

Using steve for ocpp testing and it's working really well.   Btw it's not so suitable for any external connection as no restful api are present.  i've taken a look to extension you realized but it's really custom made and i think could not be readapted to a standard scenario. 

So in the end,   plan to work on that or there's someone want to try do that with me ?

goekay, 2019-01-11:
currently there are no plans on developing restful apis for third party integration. i documented our experience and approach about how to build extensions based on steve [here](https://github.com/RWTH-i5-IDSG/steve/issues/23). i hope it helps.

Gior80, 2019-01-17:
Hi , looked at the code and seems not a big deal to work on it . You “basically” (it’s not 5 minutes 😀) extends method call from Steve internal and manage ocpp callback for response and manage communication with a dispatcher. One info : do you also extend other classes to send information about charging points to the api in some place ? (Ex get lost of all charging points with status )

goekay, 2019-01-17:
> ... extends method call from Steve internal and manage ocpp callback for response and manage communication with a dispatcher.

correct. if you have some java experience, it should be easy to work with. 

> One info : do you also extend other classes to send information about charging points to the api in some place ? (Ex get list of all charging points with status )

this is not an extension of an existing class anymore. you should write java code and classes that receive and handle such requests by reading from database.

Gior80, 2019-01-21:
clear for all,   just don't understand logic of plugsurfing, in particular how do they update list of the charger id in their db

goekay, 2019-01-21:
well, this is indeed an extension case. every time a station is [added](https://github.com/RWTH-i5-IDSG/steve-plugsurfing/blob/958b94deeb1c25edc46fbf266f699c8ab63d1480/src/main/java/de/rwth/idsg/steve/extensions/plugsurfing/controller/PlugSurfingChargePointsController.java#L77-L80) or [updated](https://github.com/RWTH-i5-IDSG/steve-plugsurfing/blob/958b94deeb1c25edc46fbf266f699c8ab63d1480/src/main/java/de/rwth/idsg/steve/extensions/plugsurfing/controller/PlugSurfingChargePointsController.java#L107-L109) using the web interface, we post it to plugsurfing.

Gior80, 2019-01-23:
thanks for all. 

meantime compiling after checking whole pom dependency still getting : 

Caused by: org.springframework.beans.BeanInstantiationException: Failed to instantiate [de.rwth.idsg.steve.config.OcppConfiguration$$EnhancerBySpringCGLIB$$48433c5e]: Constructor threw exception; nested exception is java.lang.Error: Unresolved compilation problems: 
	**ocpp cannot be resolved to a type
	ocpp cannot be resolved to a type
	CentralSystemService cannot be resolved to a type
	CentralSystemService cannot be resolved to a type
	CentralSystemService cannot be resolved to a type**

no problem at all for standard steve, i'm able to compile and run

Gior80, 2019-01-24:
Also if possible,   could i ask you,   bypassing this plugsurfing concept  ( i'm gonna make different) how the call of the remotestarttransaction is made in 3.2.0 js?  would like more to understand class and method involved in the process

goekay, 2019-01-24:
> meantime compiling after checking whole pom dependency still getting :
>
> Caused by: org.springframework.beans.BeanInstantiationException: Failed to instantiate [...]

i guess you are talking about [steve-plugsurfing](https://github.com/RWTH-i5-IDSG/steve-plugsurfing). it is not in active development anymore. so i cannot guarantee that it compiles or runs. you should take the source code only as a guideline or an inspiration for your use cases.

goekay, 2019-01-24:
> Also if possible, could i ask you, bypassing this plugsurfing concept ( i'm gonna make different) how the call of the remotestarttransaction is made in 3.2.0 js? would like more to understand class and method involved in the process

i illustrated the basic flow of an another operation (Trigger Message) in the second part of this comment: https://github.com/RWTH-i5-IDSG/steve/issues/92#issuecomment-434320972. the principles are the same for Remote Start Transaction. [here](https://github.com/RWTH-i5-IDSG/steve/blob/master/src/main/java/de/rwth/idsg/steve/web/controller/Ocpp12Controller.java#L213-L222) is your starting point.

goekay, 2019-01-24:
i consider this issue resolved and am closing it. i hope my answers were helpful.

closed 2019-01-24 by goekay