steve-community/steve #104: Database link error

sukhjitsehra, 2019-01-17:
While running mvn package i got following error:

[ERROR] Failed to execute goal org.flywaydb:flyway-maven-plugin:5.1.4:migrate (default) on project steve: org.flywaydb.core.internal.exception.FlywaySqlException: 
[ERROR] Unable to obtain connection from database (jdbc:mysql://localhost:3306/stevedb?useSSL=true&serverTimezone=UTC) for user 'steve': Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
[ERROR] ----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
[ERROR] SQL State  : 08S01
[ERROR] Error Code : 0
[ERROR] Message    : Communications link failure
[ERROR] 
[ERROR] The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.

## please suggest the issue

csamsel, 2019-01-17:
The error message is clear, the database server cannot be reached.

goekay, 2019-01-17:
@csamsel this is not his fault though. the communication between the database and java app uses SSL. depending on the used OS /JDK / mysql version, there are problems with the SSL handshake. i experienced this myself and the workaround was disabling SSL. this was not a problem for me since both steve and mysql are running in the same machine. but i was not inclined to change it in the repo, since some setups might use different machines.

@sukhjitsehra can you make the following changes and report whether it's working again:

- Set `useSSL=false` in [pom.xml](https://github.com/RWTH-i5-IDSG/steve/blob/bba63b869b592fc70ccc05d7bcc83e4410b3929b/pom.xml#L43)
- Set [this property](https://github.com/RWTH-i5-IDSG/steve/blob/bba63b869b592fc70ccc05d7bcc83e4410b3929b/src/main/java/de/rwth/idsg/steve/config/BeanConfiguration.java#L80) to false.

goekay, 2019-01-17:
excerpt from stack trace:

```
com.mysql.cj.exceptions.CJCommunicationsException: Communications link failure

The last packet sent successfully to the server was 0 milliseconds ago. The driver has not received any packets from the server.
	at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45) ~[?:?]
	at java.lang.reflect.Constructor.newInstance(Constructor.java:490) ~[?:?]
	at com.mysql.cj.exceptions.ExceptionFactory.createException(ExceptionFactory.java:61) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.exceptions.ExceptionFactory.createException(ExceptionFactory.java:105) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.exceptions.ExceptionFactory.createException(ExceptionFactory.java:151) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.exceptions.ExceptionFactory.createCommunicationsException(ExceptionFactory.java:167) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeProtocol.negotiateSSLConnection(NativeProtocol.java:351) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.negotiateSSLConnection(NativeAuthenticationProvider.java:796) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.proceedHandshakeWithPluggableAuthentication(NativeAuthenticationProvider.java:503) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.connect(NativeAuthenticationProvider.java:220) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeProtocol.connect(NativeProtocol.java:1443) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.NativeSession.connect(NativeSession.java:165) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.jdbc.ConnectionImpl.connectOneTryOnly(ConnectionImpl.java:952) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.jdbc.ConnectionImpl.createNewIO(ConnectionImpl.java:822) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	... 113 more
Caused by: javax.net.ssl.SSLHandshakeException: java.security.cert.CertPathValidatorException: Path does not chain with any of the trust anchors
	at sun.security.ssl.Alert.createSSLException(Alert.java:128) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:321) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:264) ~[?:?]
	at sun.security.ssl.TransportContext.fatal(TransportContext.java:259) ~[?:?]
	at sun.security.ssl.CertificateMessage$T12CertificateConsumer.checkServerCerts(CertificateMessage.java:642) ~[?:?]
	at sun.security.ssl.CertificateMessage$T12CertificateConsumer.onCertificate(CertificateMessage.java:461) ~[?:?]
	at sun.security.ssl.CertificateMessage$T12CertificateConsumer.consume(CertificateMessage.java:361) ~[?:?]
	at sun.security.ssl.SSLHandshake.consume(SSLHandshake.java:392) ~[?:?]
	at sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:444) ~[?:?]
	at sun.security.ssl.HandshakeContext.dispatch(HandshakeContext.java:421) ~[?:?]
	at sun.security.ssl.TransportContext.dispatch(TransportContext.java:178) ~[?:?]
	at sun.security.ssl.SSLTransport.decode(SSLTransport.java:164) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.decode(SSLSocketImpl.java:1152) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.readHandshakeRecord(SSLSocketImpl.java:1063) ~[?:?]
	at sun.security.ssl.SSLSocketImpl.startHandshake(SSLSocketImpl.java:402) ~[?:?]
	at com.mysql.cj.protocol.ExportControlled.performTlsHandshake(ExportControlled.java:315) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.StandardSocketFactory.performTlsHandshake(StandardSocketFactory.java:206) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeSocketConnection.performTlsHandshake(NativeSocketConnection.java:99) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeProtocol.negotiateSSLConnection(NativeProtocol.java:347) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.negotiateSSLConnection(NativeAuthenticationProvider.java:796) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.proceedHandshakeWithPluggableAuthentication(NativeAuthenticationProvider.java:503) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeAuthenticationProvider.connect(NativeAuthenticationProvider.java:220) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.protocol.a.NativeProtocol.connect(NativeProtocol.java:1443) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.NativeSession.connect(NativeSession.java:165) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.jdbc.ConnectionImpl.connectOneTryOnly(ConnectionImpl.java:952) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	at com.mysql.cj.jdbc.ConnectionImpl.createNewIO(ConnectionImpl.java:822) ~[mysql-connector-java-8.0.12.jar:8.0.12]
	... 113 more

```

csamsel, 2019-01-17:
Error sounds like the system wide cert store is not available. That shouldn't happen on Linux or Windows.

goekay, 2019-01-17:
new mysql jdbc driver 8.0.13 solves this issue, but introduces another one as discussed in: https://github.com/brettwooldridge/HikariCP/issues/1268

i will wait until these things are completely fixed and then update the versions to solve our problem.

sukhjitsehra, 2019-01-17:
@goekay Thanks it helped.... But now it is hanging on 

SukhjitSingh:steve test$ mvn package
[INFO] Scanning for projects...
[INFO] 
[INFO] -------------------------< de.rwth.idsg:steve >-------------------------
[INFO] Building steve 3.2.0
[INFO] --------------------------------[ jar ]---------------------------------
[INFO] 
[INFO] --- maven-enforcer-plugin:3.0.0-M2:enforce (enforce-java) @ steve ---
[INFO] 
[INFO] --- git-commit-id-plugin:2.2.4:revision (default) @ steve ---
[INFO] 
[INFO] --- properties-maven-plugin:1.0.0:read-project-properties (default) @ steve ---
[INFO] 
[INFO] --- flyway-maven-plugin:5.1.4:migrate (default) @ steve ---
[INFO] Flyway Community Edition 5.1.4 by Boxfuse
[INFO] Database: jdbc:mysql://localhost:3306/stevedb (MySQL 8.0)
[INFO] Successfully validated 31 migrations (execution time 00:00.053s)
[INFO] Creating Schema History table: `stevedb`.`schema_version`
[INFO] Current version of schema `stevedb`: << Empty Schema >>
[WARNING] outOfOrder mode is active. Migration of schema `stevedb` may not be reproducible.
[INFO] Migrating schema `stevedb` to version 0.6.6 - inital
[INFO] Migrating schema `stevedb` to version 0.6.7 - update
[INFO] Migrating schema `stevedb` to version 0.6.8 - update
[INFO] Migrating schema `stevedb` to version 0.6.9 - update
[INFO] Migrating schema `stevedb` to version 0.7.0 - update
[INFO] Migrating schema `stevedb` to version 0.7.1 - update
[INFO] Migrating schema `stevedb` to version 0.7.2 - update
[INFO] Migrating schema `stevedb` to version 0.7.3 - update
[INFO] Migrating schema `stevedb` to version 0.7.6 - update
[INFO] Migrating schema `stevedb` to version 0.7.7 - update
[INFO] Migrating schema `stevedb` to version 0.7.8 - update
[INFO] Migrating schema `stevedb` to version 0.7.9 - update
[INFO] Migrating schema `stevedb` to version 0.8.0 - update
[INFO] Migrating schema `stevedb` to version 0.8.1 - update
[INFO] Migrating schema `stevedb` to version 0.8.2 - update
[INFO] Migrating schema `stevedb` to version 0.8.4 - update
[INFO] Migrating schema `stevedb` to version 0.8.5 - update
[INFO] Migrating schema `stevedb` to version 0.8.6 - update

### it is not able to proceed to create a jar file ....

sukhjitsehra, 2019-01-17:
@goekay Thanks it helped.... But now it is hanging on

In debug mode I found that the process halt at -
[DEBUG] Executing SQL: ALTER TABLE charge_box CONVERT TO CHARACTER SET utf8 COLLATE utf8_unicode_ci

Also I was unble to execute -

mysql> GRANT SELECT ON mysql.proc TO 'steve'@'localhost';
ERROR 1146 (42S02): Table 'mysql.proc' doesn't exist

Are both connected....

goekay, 2019-01-17:
mysql made some breaking changes with version 8. steve does not support this version yet. it was reported before in #67. i need to invest some time to look at the changes and migrate the codebase to make steve compatible with mysql 8.0. for the time being, if it is possible, please use 5.7.x or mariadb.

sukhjitsehra, 2019-01-20:
@goekay It worked.

Now next step halts... sorry for that I tried other methods before posting here. I got the following issue now.

> java -jar target/steve.jar
WARNING: sun.reflect.Reflection.getCallerClass is not supported. This will impact performance.
Log file: /Users/test/logs/steve.log
Starting...WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.springframework.cglib.core.ReflectUtils$1 (file:/Users/test/Downloads/steve/target/steve.jar) to method java.lang.ClassLoader.defineClass(java.lang.String,byte[],int,int,java.security.ProtectionDomain)
WARNING: Please consider reporting this to the maintainers of org.springframework.cglib.core.ReflectUtils$1
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
........... FAILED!
Please refer to the log file for details

goekay, 2019-01-20:
the warnings that you've sent are expected. we get them as well, they are just warnings. they are because of api changes in recent jdk versions and library developers not updating their libraries to use new/legal apis. the actual problem why steve fails should be logged in `/Users/test/logs/steve.log`. please take a look at it.

i have a suspicion, that it might be because of different time zones that steve and mysql use.

goekay, 2019-05-24:
> new mysql jdbc driver 8.0.13 solves this issue, but introduces another one ...

the new version 8.0.16 fixes both the ssl issue and exceptions. i am updating the version atm.

commit 86ec716e (2019-05-24): update mysql jdbc (Closes #104)
closed 2019-05-24 by goekay