steve-community/steve #106: Fedora LXDE: Conflict between JDK of Maven and your jdk

seppy87, 2019-01-17:
Hallo,

I am using Fedora (debian isos are rejected by Virtualbox [no bootable medium found]) with LXDE. I have installed MariaDB, OpenJDK and Maven. Maven wanted to install OpenJDK 1.8 . So both are installed. This is the problem it only detects 1.8. So it says no suitable JDK is found.

`[WARNING] Rule 0: org.apache.maven.plugins.enforcer.RequireJavaVersion failed with message:
Detected JDK Version: 1.8.0-191 is not in the allowed range [11,).
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 3.794 s
[INFO] Finished at: 2019-01-17T13:00:08+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-enforcer-plugin:3.0.0-M2:enforce (enforce-java) on project steve: Some Enforcer rules have failed. Look above for specific messages explaining why the rule failed. -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[root@localhost steve-steve-3.2.0]#` 

now List of Installed JDKs
`[root@localhost steve-steve-3.2.0]# dnf list installed | grep "jdk"
copy-jdk-configs.noarch                        3.7-2.fc29                          @fedora
java-1.8.0-openjdk.x86_64                      1:1.8.0.191.b12-11.fc29             @updates
java-1.8.0-openjdk-devel.x86_64                1:1.8.0.191.b12-11.fc29             @updates
java-1.8.0-openjdk-headless.x86_64             1:1.8.0.191.b12-11.fc29             @updates
java-openjdk.x86_64                            1:11.0.1.13-11.rolling.fc29         @updates
java-openjdk-headless.x86_64                   1:11.0.1.13-11.rolling.fc29         @updates
[root@localhost steve-steve-3.2.0]#`

Any help to resolve this problem. I can't remove jdk 1.8 since it wants to remove maven too.

Greetings

seppy87, 2019-01-17:
I found a way by downloading maven from site and not using dnf install maven. I removed open-jdk 1.8 before.

goekay, 2019-01-17:
so please close this issue if it is solved.

seppy87, 2019-01-17:
how to close one?

goekay, 2019-01-17:
there is button at the bottom =D

closed 2019-01-17 by goekay