steve-community/steve #110: code clarification

Gior80, 2019-01-29:
hi guys, 

I'm making my own rest api to call steve from external.   looking at code from the plugsurfing implementation seems that chargeboxid is not  an argument from the  json  body passed to the rest but they work with a specific connectorID, (i suppose specific to their platform) and replicated on the db, from where then , using the extended "ExternalChargepointSelect" class, the chargebox id is generated and  the startsession called.  Do i have clearly understand the code? 

i'm speaking in particular about the Externalchargepointselect class extending chargepoint select. As my intention is to  pass the chargeboxid as  json argument i would use the chargepointselect class without exteding it.   am i missing something or i got it right?

     resourceimpl.java

        int connectorPK = Integer.valueOf(request.getConnectorPrimaryKey());
        ExternalChargePointSelect selectInfo = getExternalChargePointSelect(connectorPK);

        switch (selectInfo.getVersion()) {
            case V_12:
                ocpp12Mediator.processStartTransaction(rfid, selectInfo, wrapper);
                break;
            case V_15:
                ocpp15Mediator.processStartTransaction(rfid, selectInfo, wrapper);
                break;
        }
    }

goekay, 2019-01-30:
> looking at code from the plugsurfing implementation seems that chargeboxid is not an argument from the json body passed to the rest but they work with a specific connectorID

correct. in steve's db we have the unique tuples `(charge box id, connector id, connector pk)`. `connector pk` is the auto-generated db primary key for the unique 2‑tuple `(charge box id, connector id)`. this 2‑tuple is part of the ocpp spec. since plugsurfing does not care for charging _stations_ but charging _points_, we decided to expose `connector pk` to uniquely identify charging points. the confusing part is that in the json message this field is called `connector-id`. all in all, the mapping is: `connector pk` (of steve's db) = `connector-id` (of plugsurfing)

> ... the chargebox id is generated ...

it is _not_ generated. it is looked up in the database.

Gior80, 2019-02-01:
Got it ! So basically to call the start transaction, remote system need to know pk of Steve. This mean I need to replicate the pk table on remote system  , right ? If this is the case Then same is needed for users and tag and again for charging points status . 

Basic problem I see here (that could run in desync problem if connection between 2 system is lost) is to have   synchronized Local and remote platform on all those stuff. Just to think at best approach in doing my project , can I ask if you maybe have already think about  this in your  consideration? Ex: if a new chargepoint is added on Steve , then “moved” to ps but there’s no connection and client call fail, then this charging point is lost forever on ps? 

Thanks again for super work and support

goekay, 2019-02-01:
> remote system need to know pk of Steve. This mean I need to replicate the pk table on remote system , right ? If this is the case Then same is needed for users and tag and again for charging points status .

correct. this is also what steve-plugsurfing does. every time charging station information is changed (with respect to connectors or status of station/connectors) steve-plugsurfing notifies and updates the remote system.

the handling of users is different. steve-plugsurfing allows every remote plugsurfing user to charge. in this sense, if it the user exists in database, perfect. otherwise, we insert the user into the database during the processing of start transaction message.

>  Ex: if a new chargepoint is added on steve-plugsurfing, then “moved” to ps but there’s no connection and client call fail, then this charging point is lost forever on ps?

not necessarily. there are two fallbacks as best practice to prevent this (but we did not implement these in steve-plugsurfing. in our case _if it failed, it failed_):
- some retry logic with backoff, where steve is the active communicator. more info: https://stackoverflow.com/questions/32352484/retry-java-resttemplate-http-request-if-host-offline
- the retry logic will give up at some point. therefore additionally the remote system can be active and refresh its database by asking steve about all stations/connectors in fixed intervals, e.g. every night or weekly. in this solution, the station will _eventually_ appear in remote system. the sooner the better, therefore the decision of fixed interval depends on your use case.

Gior80, 2019-02-06:
hi goekay, 

still going on in  my extensions and got stuck on a point about managing users. 
Coming from your previous indication:

> the handling of users is different. steve-plugsurfing allows every remote plugsurfing user to charge. in this sense, if it the user exists in database, perfect. otherwise, we insert the user into the database during the processing of start transaction message

taking the sessionStart

> private void sessionStart(AsyncHttpWrapper wrapper, SessionStart request) {
        validate(request);
        String rfid = getRfid(request.getUser());

i have couple of things confusing me. First looking at sessionstart 

    @NotNull(message = "user is missing")
    private User user;
    @NotEmpty(message = "connector-id is missing")
    @JsonProperty(value = "connector-id")
    private String connectorPrimaryKey;
    private String paymentReference; 

i'm probably wrong but is not the connectorid the only deserialized (  @JsonProperty ) parameter on request ? should the jsonproperty not be declared also on other variables?  how the json you're sending is made? (sorry for 1 million question 😃 ) looks like  "user" represent the rfid in real world so  not a  user name.  Seems also private token in User DTO remain unused 😕 , i'm right? 

so this is ipotetic json i reconstructed
{
  "session-start": {
  	"user":"Mrwhite",   <- IS THIS RFID IN REAL WORLD???
    "connector-id":"12",
    "paymentReference":"12345"
    }
}
 
coming to that ,  how and where the user existance check / creation is made?   coming to our first assumption that anyone can start/stop session i'm confused on how to manage that. The point of sending the rfid code would mean that , think at a situation where user has just the app and not card, i need to create a "dummy" rfid code to manage this process.  This could obviously work but is to understand. 

thanks again

goekay, 2019-02-07:
hello, i want to help anyone who uses steve and wants to build something cool on top of steve, but this chain of questions is getting out of hand. i cannot help you at each step you take. your questions can be answered by: _please read the code and follow the method calls_. and if you are somewhat familiar with java, web technologies and the libraries/frameworks we use, extending steve should not be hard. and if you are not, github issues is not the place learn them. i will give you one last hint to overcome the problem you face, but please continue on your own. i ask for your understanding.

`private User user` is not a simple field but a complex object, which contains other fields. depending on its `identifierType` we take different actions.

closed 2019-02-07 by goekay

Gior80, 2019-02-07:
Hello ,

I understand the pos and meantime let me say  thanks for help till now