steve-community/steve #111: callback of task

Gior80, 2019-02-11:
Hello,

i've been able to create a standard api using your last indication.  I also extended the database over a google firestore instance.  everything work like charme so thanks for now!  
I would like to ask you if you can exaplain me more about how to get the calback from the task.
it's really easy to launch the task similar way of the web request.

`CommunicationTask i = requestTaskStore.get(client16.remoteStartTransaction(params)); `

this way i can get the task running and get his instance. Message is sent to chargepoint and i got response, all db table are updated.  I tried at this point to  add a callback to the task to manage api response usig the AsyncWrapper.  I  created the callback
`
OcppCallback<RemoteStartTransactionResponse>
 cb = new OcppCallback<RemoteStartTransactionResponse>() {
//implements success and fails 
}
`
add then the ref callback to communicationTask Instance i got from the client

`i.addCallback(mycallback);`

i never get a  callback  in the cb callback i created. Would you have suggestion ?  do i am working in on wrong area ?  the defaultOcppCallBack in CommunicationTask is instead working.

Thanks

goekay, 2019-02-11:
github issues are not a support channel. i want to remind you of my last response: https://github.com/RWTH-i5-IDSG/steve/issues/110#issuecomment-461357563

regarding your problem (this is my last bit of help to you): you should not get the task from `requestTaskStore` and modify it by adding callbacks. at this point, it is probably too late, because the response already arrived and all registered callbacks are executed when processing the response. you should register the callbacks at the place where you initialize the task, before the request is sent. not later.

please do not open any more issues about your extension.

closed 2019-02-11 by goekay

Gior80, 2019-02-11:
hi goakey, 

I absolutely remeber. btw have no other way to contact you except of here.   As seems you encourage and support forks and consider i was also doing that  for releasing  to other (as seems other was looing to extend)  could a different communication channel be better to speak?
thanks