steve-community/steve #112: Problem with installation

michalmalekci, 2019-02-11:
Hi
I'm trying to build the app, but I receive an error
It looks like the app tries to migrate from the previous version (db). 

[steve.log](https://github.com/RWTH-i5-IDSG/steve/files/2853100/steve.log)

goekay, 2019-02-11:
hi, well.. the error is kinda self-explanatory. the database user that you use for steve does not have the necessary privileges to execute migration scripts. 0.6.6 is the first version we started using flyway with ([migrations](https://github.com/RWTH-i5-IDSG/steve/tree/master/src/main/resources/db/migration)). make sure that you follow the database preparation steps from README.

one other thing: it looks like you are using MySQL 5.5. you should use version 5.7.7 or later. this is mentioned in README as well.

michalmalekci, 2019-02-11:
Hi
I use MariaDB 10.3
I created the schema and the user with the privileges.

goekay, 2019-02-13:
i could reproduce the problem under ubuntu 18.10 with mariadb 10.3.12. after googling a bit, i found this solution which worked for me: http://wiki.ispirer.com/sqlways/troubleshooting-guide/mysql/import/binary-logging

the database triggers were used before, but we removed them with newer migrations. therefore this setting is only necessary to apply existing/old migrations and you can revert this setting back after you executed them.

closed 2019-02-13 by goekay